FreeRTOS/FreeRTOS-Plus-TCP #40: CRC Error false alarm in Zynq driver

siddhantmodi, 2020-10-20:
Based on [this](https://forums.freertos.org/t/freertos-tcp-ip-on-arm-cortexa53/9898/50?u=sidmodi) forum comment from Hein, the following change needs to be made in Zynq/NetworkInterface.c around line 234:

`-    if( ( pxPacket->xICMPPacket.xIPHeader.ucProtocol != ipPROTOCOL_UDP ) &&
+    if( ( pxPacket->xICMPPacket.xEthernetHeader.usFrameType == ipIPv4_FRAME_TYPE ) &&
+        ( pxPacket->xICMPPacket.xIPHeader.ucProtocol != ipPROTOCOL_UDP ) &&
         ( pxPacket->xICMPPacket.xIPHeader.ucProtocol != ipPROTOCOL_TCP ) )
     {
         /* The EMAC will calculate the checksum of the IP-header.
         It can only calculate protocol checksums of UDP and TCP,
         so for ICMP and other protocols it must be done manually. */
         usGenerateProtocolChecksum( (uint8_t*)&( pxPacket->xUDPPacket ), pxBuffer->xDataLength, pdTRUE );
     }`

Without that change, the message `CRC error: 1234 location 3` is printed in the serial console whenever the device send an ARP message out.

siddhantmodi, 2020-10-20:
I just saw that a PR is in progress for this. Closing issue.

closed 2020-10-20 by siddhantmodi

AniruddhaKanhere, 2020-10-20:
Hello @siddhantmodi, thanks for investing your time and energy in creating this PR.
Yes, there is a PR open and we are working on merging that to the mainline.