FreeRTOS/FreeRTOS-Plus-TCP #160: FreeRTOS_DHCP.c missing socket include

arturlangner, 2021-01-18:
FreeRTOS_DHCP.h is missing `#include "FreeRTOS_Sockets.h"`. Build fails at `BaseType_t xIsDHCPSocket( Socket_t xSocket );` https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/09192b0e43998c8aeb0654fd0ff5bf3b102b78c9/include/FreeRTOS_DHCP.h#L104

AniruddhaKanhere, 2021-01-20:
Hello @arturlangner,

Thanks for taking the time to create an issue for this.

May I ask what build system are you using for compiling the project?
I ask this because the build system I am using - cmake and Visual studio, is able to compile without any errors.

RichardBarry, 2021-01-20:
Would also be interested in which .c file generates the error as it is probably something to do with the order of includes.  Your FreeRTOSIPConfig.h file might be of interest too if that is changing what is and isn't included in the build.

arturlangner, 2021-01-20:
The offending .c file was actually my own. I swapped the lines with `#include <FreeRTOS_Sockets.h>` and `#include <FreeRTOS_DHCP.h>`. Builds fine with gcc.

closed 2021-01-20 by arturlangner