FreeRTOS/FreeRTOS-Plus-TCP #212: fatal error: esp_wifi_internal.h: No such file or directory

hackaprende, 2021-03-17:
In the new FreeRTOS version I have this issue

fatal error: esp_wifi_internal.h: No such file or directory

It did not happen in the last version. Please tell me if you need more information about my environment.

AniruddhaKanhere, 2021-03-17:
Hello @hackaprende,
Apologies for the trouble you have had.

I will require a bit more information for debugging and replicating the problem you are having. Could you let me know which project are you trying to compile - is it on GitHub or are you using a local project?
What is the build system that you are using? CMake or ESP-IDF?

hackaprende, 2021-03-18:
I solved it by using:
#include "esp_private/wifi.h"

Instead of 
#include esp_internal_wifi.h

I also shared my solution here
https://github.com/aws/amazon-freertos/issues/3053

AniruddhaKanhere, 2021-03-18:
@hackaprende, Thanks for sharing that solution here. Would you like to create a PR for the same in the TCP repository?
If not, we would gladly do it for you.

hackaprende, 2021-03-18:
Sure, no problem, I am going to create the PR today.

AniruddhaKanhere, 2021-03-31:
Hello @hackaprende, as mentioned in the PR, using the version of ESP-IDF 4.2 makes the code work for me without any change. There is quite a bit of change between version 3.3 and 4.2.
Was this error fixed once you applied the fix mentioned in this Issue: https://github.com/aws/amazon-freertos/issues/3054?