FreeRTOS/FreeRTOS-Plus-TCP #218: Unnecessary pointer leading to compiler warning on GCC

tajen, 2021-03-30:
When compiling FreeRTOS_IP.c I get the following warning:

FreeRTOS_IP.c:1949:66: warning: taking address of packed member of 'struct xUDP_HEADER' may result in an unaligned pointer value [-Waddress-of-packed-member]

also for lines: 2547, 2563 and 2580.

Here is the offending code:
```
 pxProtPack = ipCAST_PTR_TO_TYPE_PTR( ProtocolPacket_t, &( pxNetworkBuffer->pucEthernetBuffer[ uxHeaderLength - ipSIZE_OF_IPv4_HEADER ] ) );
                            pusChecksum = ( const uint16_t * ) ( &( pxProtPack->xUDPPacket.xUDPHeader.usChecksum ) );

                            if( *pusChecksum == ( uint16_t ) 0U )
                            {
                                #if ( ipconfigHAS_PRINTF != 0 )
                                    {
...
```
It may be a problem problem for 32-bit aligned memory.

Here is the struct being pointed in to:
```
struct xUDP_HEADER
    {
        uint16_t usSourcePort;      /**< The source port                      0 + 2 = 2 */
        uint16_t usDestinationPort; /**< The destination port                 2 + 2 = 4 */
        uint16_t usLength;          /**< The size of the whole UDP packet     4 + 2 = 6 */
        uint16_t usChecksum;        /**< The checksum of the whole UDP Packet 6 + 2 = 8 */
    }
```

I don't see why it is necessary to make a pointer to the value just to dereference it - instead of just accessing it directly?

AniruddhaKanhere, 2021-03-30:
Hello @tajen, thanks for bringing this to our attention. On a quick look, it seems that you are right. We are not using this for any other reason than to check its value.
Let me do some digging and figure out what is the best way to handle this.

htibosch, 2021-03-31:
[This local declaration](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/FreeRTOS_IP.c#L1962) of `pusChecksum` can be avoided easily.

The use of `pusChecksum` in the function [`usGenerateProtocolChecksum`](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/FreeRTOS_IP.c#L2496) is quite essential to the code.

As you probably know: when a `uint16_t` is declared in a packed struct, it may well be located at a non-aligned address, and dereferencing `pusChecksum` may lead to an exception.

However, by design, we know that all 2-byte and 4-byte struct members are aligned correctly.

In fact, there is one exception, which is [`ARPHeader_t::ucSenderProtocolAddress`](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/include/FreeRTOS_IP_Private.h#L104), which is declared as an array of bytes:

	uint8_t ucSenderProtocolAddress[ 4 ]; /**< Network address of sender */

I am not sure what to do about the use of `pusChecksum` in the function `usGenerateProtocolChecksum()`. Let's give it another thought. I'm afraid it needs a lot of extra coding to avoid that compiler warning. Any idea?

AniruddhaKanhere, 2021-03-31:
Hello @tajen, as Hein correctly points out, it seems that only the first occurrence that you mention should be removed and we can replace that directly with the structure.
All other occurrences are quite essential. They cannot be easily removed/replaced. I agree with Hein on the point that we should provide more thought to this before hacking away at the code. Any suggestions are welcome :)

tajen, 2021-04-06:
I see you point. 
To be honest, I only checked the first occurance. And it was in the the LTS version, so the line numbers are not correct for the main branch. I appologize for that.

I will see if I can come up with a usable solution.

AniruddhaKanhere, 2021-04-10:
Hello @tajen, no need for apologies! :)

Since there is no straightforward way to get around this particular warning, and since I do not think that we can categorize this as a bug, I do not see value for other customers in keeping this issue open.

I shall close this issue - you're welcome to reopen this issue or another one in the future - whatever suits your needs. Thanks again for bringing this to our attention.

For the first instance, you're welcome to create a PR. If not, then we can create the PR on your behalf. What do you think?

closed 2021-04-10 by AniruddhaKanhere