FreeRTOS/FreeRTOS-Plus-TCP #235: IPv6 router solicitations are sent from non link-local addresses

setdebug, 2021-05-01:
OpenBSD's rad(8) (and probably other implementations) will not respond to router solicitations sent from non link-local addresses. As a result, the endpoint falls back to a randomly assigned address when no router advertisement is received.

To work around this, I set a link-local address on the interface:

```
FreeRTOS_inet_pton6("fe80::9999", endpoint.ipv6_settings.xIPAddress.ucBytes);
endpoint_v6_slaac.bits.bWantRA = pdTRUE;
```

That works and the endpoint takes on an address with the prefix from the router advertisement message; however, after a certain period of time, the endpoint can no longer be pinged at the assigned address. This is the log output when that happens:

```
vRAProcess( 0, 0x2000cfccip) bRouterReplied=1 bIPAddressInUse=0 state 4 -> 0
vIPReloadDHCP_RATimer: 1000
vRAProcess: Router Solicitation, attempt 1/3
vRAProcess( 0, 0x2000cfccip) bRouterReplied=1 bIPAddressInUse=0 state 0 -> 1
vIPReloadDHCP_RATimer: 10000
ICMPv6 135 (NEIGHBOR_SOL) from 0x200033d0ip to 0x200033e0ip end-point = 0
vRAProcess: Router Solicitation, attempt 2/3
vRAProcess( 0, 0x2000cfccip) bRouterReplied=1 bIPAddressInUse=0 state 1 -> 1
vIPReloadDHCP_RATimer: 10000
ICMPv6 133 (ROUTER_SOL) from 0x200033d0ip to 0x200033e0ip end-point = 1
vRAProcess: Router Solicitation, attempt 3/3
vRAProcess( 0, 0x2000cfccip) bRouterReplied=1 bIPAddressInUse=0 state 1 -> 1
vIPReloadDHCP_RATimer: 10000
RA: Giving up waiting for a Router.
RA: Creating a random IP-address
RA: Neighbour solicitation for 0x2000cf78ip
vRAProcess( 0, 0x2000cfccip) bRouterReplied=0 bIPAddressInUse=0 state 1 -> 3
vIPReloadDHCP_RATimer: 1000
prvAllowIPPacketIPv6: drop 0x200033e0ip (from 0x200033d0ip)
RA: Neighbour solicitation for 0x2000cf78ip
vRAProcess( 0, 0x2000cfccip) bRouterReplied=0 bIPAddressInUse=0 state 3 -> 3
vIPReloadDHCP_RATimer: 1000
prvAllowIPPacketIPv6: drop 0x200033e0ip (from 0x200033d0ip)
RA: Neighbour solicitation for 0x2000cf78ip
vRAProcess( 0, 0x2000cfccip) bRouterReplied=0 bIPAddressInUse=0 state 3 -> 3
vIPReloadDHCP_RATimer: 1000
prvAllowIPPacketIPv6: drop 0x200033e0ip (from 0x200033d0ip)
RA: Neighbour solicitation for 0x2000cf78ip
vRAProcess( 0, 0x2000cfccip) bRouterReplied=0 bIPAddressInUse=0 state 3 -> 3
vIPReloadDHCP_RATimer: 1000
RA: failed, using default parameters and IP address 0x2000cf78ip
```

I think it's trying to request another RA but it's doing so from the previously-assigned non link-local address (and rad is refusing to send anything). It times out and sets the address to something random.

A possible solution might be to allow the user to specify a separate endpoint or source address to use for sending router solicitations. Another idea is using `ipv6_defaults.xIPAddress` instead of `ipv6_settings.xIPAddress` and having a link-local address assigned as the default address.

(in FreeRTOS_RA.c, line 125):
```
( void ) memcpy( pxICMPPacket->xIPHeader.xSourceAddress.ucBytes, pxEndPoint->ipv6_defaults.xIPAddress.ucBytes, 16 );
```

Or maybe there should always be an implicit link-local endpoint present when there are IPv6 endpoints.

htibosch, 2021-05-01:
@setdebug Thanks for reporting this. I am glad to see that users start adopting IPv6.

When you have more questions about the [ipv6_multi branch](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/labs/ipv6_multi), maybe the [FreeRTOS Forum](https://forums.freertos.org) is a better place, it will have more viewers.

First a remark about the logging of IP-addresses. I normally link my projects with this [printf-stdarg.c](https://github.com/htibosch/freertos_plus_projects/blob/master/plus/Common/Utilities/printf-stdarg.c).

It adds two new printf formats: `%xip` ( for "192.168.2.1" ), and `%pip` ( for "fe80::9999" ).

In my setup I have the rad(8) running on a Raspberry pie. That device also serves as an IPv6 router, using a 6to4 tunnel.

I just ran my testing device two times:

router_on.log : while my raspberry was serving as a router
router_off.log : while my raspberry was off

Please find the logging here: [logging_ra.zip](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/6409127/logging_ra.zip)

Your logging looks like my "router_off.log" :

RA: Giving up waiting for a Router.
RA: Creating a random IP-address
RA: Neighbour solicitation for 2001:470:ec...

where "2001:470:ec..." is my IPv6 network.

The idea is as follows:
1. Issue a Router Solicitation ( at most three times )
2. If answered: create a random address and check if it already exists
3. If there is no Router Advertisement: create a random address base on a known network address ( taken from `ipv6_defaults` ).

Here you see the ICMPv6 messages during a successful startup:

[ra_conversation.zip](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/6409148/ra_conversation.zip)

Here is the initialisation code:
~~~c
IPv6_Address_t xIPAddress;
IPv6_Address_t xPrefix;
IPv6_Address_t xGateWay;
IPv6_Address_t xDNSServer;

/* The default IPv6 network. */
FreeRTOS_inet_pton6( "2001:470:ec54::",           xPrefix.ucBytes );

/* A DNS address. The IPv6 DNS addresses are not yet in use by FreeRTOS_DNS.c */
FreeRTOS_inet_pton6( "2001:4860:4860::8888",      xDNSServer.ucBytes );            

/* Create a randomised IPv6 address, based on the network. */
FreeRTOS_CreateIPv6Address( &xIPAddress, &xPrefix, 64, pdTRUE );

/* And a gateway address. */
FreeRTOS_inet_pton6( "fe80::9355:69c7:585a:afe7", xGateWay.ucBytes );

FreeRTOS_FillEndPoint_IPv6( &( xInterfaces[ 0 ] ), 
                            &( xEndPoints[ 2 ] ),
                            &( xIPAddress ),
                            &( xPrefix ),
                            64uL,                /* Prefix length. */
                            &( xGateWay ),
                            &( xDNSServer ),    /* pxDNSServerAddress: Not used yet. */
                            ucMACAddress1 );
#if( ipconfigUSE_RA != 0 )
{
    pxEndPoint_2->bits.bWantRA = pdTRUE;
}
#endif /* #if( ipconfigUSE_RA != 0 ) */
~~~

AniruddhaKanhere, 2021-06-09:
@setdebug, is this issue resolved? Or do you need more clarification/help?

setdebug, 2021-07-03:
@AniruddhaKanhere and @htibosch:

Sorry for not getting back to you guys. I created this issue because there doesn't appear to be a way to ensure that router solicitations are always sent from a link-local address (unless I am missing something). If I create an IPv6 endpoint with `bWantRA` enabled, it is assigned an address the first time (it works) but subsequent solicitations/renewals are ignored by the router because they are coming from the endpoint's address rather than a link-local address. Looking at [RFC 4861](https://datatracker.ietf.org/doc/html/rfc4861#section-6.1.1), RAs are required to come from the router's link-local address. I don't see an explicit requirement that solicitations come from link-local addresses; however, I know for sure that rad(8) will ignore them if they don't.

I think there needs to be some way to specify the source address for router solicitations. I provided a few ideas in my first comment but I don't know if they are adequate. Using `ipv6_defaults.xIPAddress` as the source address would be an easy (possibly temporary) fix if it is assigned an `fe80::` address. I would like it if I didn't have to modify FreeRTOS+TCP code.

htibosch, 2021-07-04:
I was thinking along the same line, use `ipv6_defaults.xIPAddress` as the source address in the RA message.
I will prepare and test this option.
Thanks!

htibosch, 2021-07-04:
The only problem with assigning a link-local address to `ipv6_defaults`, occurs when the router is out-of-service. I would still like to give it a public IP-address.
( when a DHCP server does not respond at start-up, many device also use the last used IP-address ).
Also, I would like to give a single link-local address to an interface. I normally have 2 IPv6 endpoints on a single interface:

- A link-local address, easy to remember: fe80::7007
- A public IP-address, using my (fixed) IPv6 prefix, e.g.: 2001:407:ce54::???

So I wonder if we can leave it the way it is, and just make sure that the return address in the RA solicitation is the local-link address of the interface.

htibosch, 2021-07-29:
Hi @setdebug, have you tried the solution as provided in PR #271 ?

AniruddhaKanhere, 2021-08-05:
Hello @setdebug, I am resolving this issue with a hope that the solution mentioned by @htibosch worked for you.
If you have any more questions/concerns, feel free to reopen this issue or open a new one.

Thanks once again for bringing this to our attention.

closed 2021-08-05 by AniruddhaKanhere