FreeRTOS/FreeRTOS-Plus-TCP #251: Unknown type name 'SocketSet_t' / 'Socket_t'

tieugiang91, 2021-05-12:
I'm trying to cross compile a library using FreeRTOS and toolchain file as:

```
include(CMakeForceCompiler)

set(CMAKE_SYSTEM_NAME Generic)
set(PLATFORM_NAME "FreeRTOS_Plus_TCP")
set(CMAKE_CROSSCOMPILING True)

set(CMAKE_C_COMPILER xtensa-lx106-elf-gcc)
set(CMAKE_CXX_COMPILER xtensa-lx106-elf-g++)
set(CMAKE_ASM_COMPILER xtensa-lx106-elf-gcc)

set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY)

set(FREE_RTOS_DIR $ENV{IDF_PATH}/components/freertos)
set(ESP8266_DIR $ENV{IDF_PATH}/components/esp8266)
include_directories(SYSTEM
	${FREE_RTOS_DIR}/include
	${FREE_RTOS_DIR}/include/freertos
	${FREE_RTOS_DIR}/include/freertos/private
	${FREE_RTOS_DIR}/port/esp8266/include
	${FREE_RTOS_DIR}/port/esp8266/include/freertos
	$ENV{IDF_PATH}/components/heap/include
	$ENV{IDF_PATH}/components/heap/port/esp8266/include
	${ESP8266_DIR}/include
	$ENV{IDF_PATH}/components/esp_common/include
	/home/tyna/Downloads/FreeRTOSv202012.00/FreeRTOS-Plus/Source/FreeRTOS-Plus-TCP/include
	/home/tyna/Downloads/FreeRTOSv202012.00/FreeRTOS-Plus/Source/FreeRTOS-Plus-TCP/portable/Compiler/GCC
)
```
The compilation gave the error:
Unknown type name 'SocketSet_t' / 'Socket_t'
Which header should I include for Socket_t and SocketSet_t please? Thank you.

htibosch, 2021-05-13:
I would recommend to include the following header files:
~~~c
/* FreeRTOS includes. */
#include <FreeRTOS.h>

/* FreeRTOS+TCP includes. */
#include "FreeRTOS_IP.h"
#include "FreeRTOS_Sockets.h"
~~~
And in case you need socket sets ( `SocketSet_t` ), your copy of FreeRTOS_IPConfig.h needs to contain the following define:
~~~
#define ipconfigSUPPORT_SELECT_FUNCTION   1
~~~

tieugiang91, 2021-05-13:
Thank you, finally it works.

closed 2021-05-13 by tieugiang91

htibosch, 2021-05-13:
The next time when you have a question, I recommend using the [FreeRTOS forum](https://forums.freertos.org). It is attended by many users and professionals who can answer your questions.
This issue section on github is more for bug-reports. It is read by few people.