FreeRTOS/FreeRTOS-Plus-TCP #275: FreeRTOS_IPInit always reports failure when STATIC_ALLOCATION=1

sefthyvoulos, 2021-07-19:
FreeRTOS_IPInit always reports failure when configSUPPORT_STATIC_ALLOCATION=1. Looks like in this path the return value isn't updated to reflect success.

AniruddhaKanhere, 2021-07-20:
Hello @sefthyvoulos, thank you for raising the issue and letting us know about this.

Would you mind letting us know which version of the FreeRTOS+TCP and the FreeRTOS-Kernel are you using? We are looking into this issue. Thank you for your patience.

htibosch, 2021-07-21:
It looks like the function `FreeRTOS_IPInit()` forgets to set the return value `xReturn` in case static allocation is enabled.
I will propose the following change to FreeRTOS_IP.c":
~~~c
 /* Create the task that processes Ethernet and stack events. */
 #if ( configSUPPORT_STATIC_ALLOCATION == 1 )
     {
         static StaticTask_t xIPTaskBuffer;
         static StackType_t xIPTaskStack[ ipconfigIP_TASK_STACK_SIZE_WORDS ];
         xIPTaskHandle = xTaskCreateStatic( prvIPTask,
                                            "IP-Task",
                                            ipconfigIP_TASK_STACK_SIZE_WORDS,
                                            NULL,
                                            ipconfigIP_TASK_PRIORITY,
                                            xIPTaskStack,
                                            &xIPTaskBuffer );
+        if( xIPTaskHandle != NULL )
+        {
+            xReturn = pdTRUE;
+        }
     }
 #else /* if ( configSUPPORT_STATIC_ALLOCATION == 1 ) */
~~~
I will prepare a small PR for this, thanks for reporting!

AniruddhaKanhere, 2021-07-21:
The above PR was merged. Thank you @htibosch for making that change.
Thank you @sefthyvoulos for pointing this out. I shall be closing this issue now.

If you have any other issue/problem, feel free to open another issue or reopen this issue.

closed 2021-07-21 by AniruddhaKanhere