FreeRTOS/FreeRTOS-Plus-TCP #292: Link-Local Multicast Name Resolution issue when using BufferAllocation_1 is used

JeroenWagenaar, 2021-08-01:
When BufferAllocation_1 is used the buffers are allocated fixed size. FreeRTOS_DNS.c does not support this. pucNewBuffer is initialized to NULL. 

fix at line 1503

`                            else if( ( xBufferAllocFixedSize == pdTRUE ) && ( pxNetworkBuffer != NULL ) )
                            {
                            	pucNewBuffer = &( pxNetworkBuffer->pucEthernetBuffer[ ipUDP_PAYLOAD_OFFSET_IPv4 ] );
                            }
`

htibosch, 2021-08-02:
Thanks a lot for reporting, Jeroen!
Before your patch:
~~~bash
C:\>ping stm32f4
Ping request could not find host stm32f4. Please check the name and try again.
~~~
After applying the patch:
~~~bash
C:\>ping stm32f4

Pinging stm32f4 [192.168.2.13] with 32 bytes of data:
Reply from 192.168.2.13: bytes=32 time=1ms TTL=128
Reply from 192.168.2.13: bytes=32 time<1ms TTL=128
Reply from 192.168.2.13: bytes=32 time<1ms TTL=128
~~~
LLMNR used to work well, I used it in many projects, but during some PR the essential assignment must have disappeared.

If you like you can prepare a PR ( "pull request" ) for this. In case you're not experienced, we will guide you through.
If not, I don't mind creating a PR my self.
The code can also be put like this:
~~~c
if( pxNetworkBuffer != NULL )
{
    if( xBufferAllocFixedSize == pdFALSE )
    {
        /* pxNetworkBuffer might be come NULL here. */
    }
    else
    {
        pucNewBuffer = &( pxNetworkBuffer->pucEthernetBuffer[ ipUDP_PAYLOAD_OFFSET_IPv4 ] );
    }
}

if( ( pxNetworkBuffer != NULL ) && ( pucNewBuffer != NULL ) )
~~~

Background: the reply to an LLMNR message is longer that the request. Now in case the network buffers have a variable size (`xBufferAllocFixedSize=false`), a new bigger buffer must be allocated.
When 'BufferAllocation_1.c' is used, the incoming buffer can be reused, because it has a fixed size of `ipconfigNETWORK_MTU + ipSIZE_OF_ETH_HEADER` (14), usually 1514 bytes.

htibosch, 2021-08-03:
A few hints, in case you need that:

- The PR is supposed to be based on the "main" branch
- Please add some text about the why and how
- Before submitting, apply [uncrustify](https://sourceforge.net/projects/uncrustify/files/uncrustify/uncrustify-0.66.1) to the changed source code, using [this uncrustify.cfg](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/tools/uncrustify.cfg).

If you don't manage to get the source "uncrustified", it can be done later.

Thanks, Hein

closed 2021-08-08 by JeroenWagenaar