FreeRTOS/FreeRTOS-Plus-TCP #380: Rename EMAC stack size variable and document

rpcme, 2021-11-16:
The define `configEMAC_TASK_STACK_SIZE` in NetworkInterface.c infers this is a FreeRTOS kernel related configuration when I feel that it should have configip

Besides this it is an undocumented configurable stack size.  I faced stack overflow in the EMAC task, then defined it in FreeRTOSIPConfig.h with increase, and all was good.

I would also expect it to be documented here.

https://freertos.org/FreeRTOS-Plus/FreeRTOS_Plus_TCP/TCP_IP_Configuration.html﻿

htibosch, 2021-11-19:
The macro `configEMAC_TASK_STACK_SIZE` is used in 12 versions of NetworkInterface.c.

What about the following new code:
~~~c
#ifndef ipconfigEMAC_TASK_STACK_SIZE_WORDS
    /* Check if the earlier name of the macro is still defined. */
    #ifdef configEMAC_TASK_STACK_SIZE
        #warning This macro is now called ipconfigEMAC_TASK_STACK_SIZE_WORDS
        #define ipconfigEMAC_TASK_STACK_SIZE_WORDS   ( configEMAC_TASK_STACK_SIZE )
    #else
        #define ipconfigEMAC_TASK_STACK_SIZE_WORDS   ( 2U * configMINIMAL_STACK_SIZE )
    #endif
#endif
~~~

@rpcme wrote:

> I would also expect it to be documented here.
> 
> https://freertos.org/FreeRTOS-Plus/FreeRTOS_Plus_TCP/TCP_IP_Configuration.html

FreeRTOS+TCP has a dozen of configuration macro's that get their default value within one of the source files. This is an example of such a macro:
~~~c
/** @brief Time delay between repeated attempts to initialise the network hardware. */
#ifndef ipINITIALISATION_RETRY_DELAY
    #define ipINITIALISATION_RETRY_DELAY    ( pdMS_TO_TICKS( 3000U ) )
#endif
~~~
It is defined there locally, and it is not mentioned in `FreeRTOSIPConfigDefaults.h` or in the documentation. The reason for this hiding is that the defaults work well for almost everyone.

I think it is good to change the status of `ipconfigEMAC_TASK_STACK_SIZE_WORDS`: make it public, and add it to the public documentation and also `FreeRTOSIPConfigDefaults.h`.

Agreed?

htibosch, 2022-01-26:
> Would you like to create a PR for this?

Sure I will. I silently thought that this is a PR in stead of an issue.

moninom1, 2023-07-23:
Closing for now as stack size is different for all network interfaces.

closed 2023-07-23 by moninom1