FreeRTOS/FreeRTOS-Plus-TCP #381: Document required call in STM32F7 port readme

rpcme, 2021-11-16:
Document in STM32F7 port readme that `void HAL_ETH_MspInit(ETH_HandleTypeDef *heth)﻿` implementation by user is required.

Other drivers that I have seen like lwip provide the pin.

I assumed that this would have been covered by the STM32F746G BSP or the provided FreeRTOS+TCP driver.

htibosch, 2021-11-19:
Thanks @rpcme for this suggestion.

What about adding the following text to readme.txt :

---
When initialising the EMAC, the driver will call the function `HAL_ETH_MspInit()`, which should do the following:

- Enable the Ethernet interface clock using:
    __HAL_RCC_ETHMAC_CLK_ENABLE();
    __HAL_RCC_ETHMACTX_CLK_ENABLE();
    __HAL_RCC_ETHMACRX_CLK_ENABLE();

- Initialize the related GPIO clocks

- Configure Ethernet pin-out
    Please check the scheme of your hardware to see which pins are used.
    Also check if either MII or RMII is used ( define `ipconfigUSE_RMII`
    as 0 or 1 ).

- Configure Ethernet NVIC interrupt (IT mode)
    Choose a proper interrupt priority, defined in FreeRTOSIPConfig.h as e.g. :
~~~c
   `#define ipconfigMAC_INTERRUPT_PRIORITY	( configLIBRARY_MAX_SYSCALL_INTERRUPT_PRIORITY )
~~~

The function `HAL_ETH_MspInit()` must be provided by the application. Make sure that your copy of the function is called, and not a dummy defined as "weak".

AniruddhaKanhere, 2022-01-26:
@htibosch, would you like to create a PR to add that in the readme?

AniruddhaKanhere, 2022-03-15:
Closing this issue as the readme was updated with PR 434