FreeRTOS/FreeRTOS-Plus-TCP #390: Compilation failure in DNS source file

go2sh, 2021-12-06:
I downloaded the latest multi/IPv6 branch and compiled it in default configuration. The DNS code produces compilations errors:

- pcName is disabled in ParseSet_t, even needed for dns operation
- prvReadNameField is disabled, even needed
- ucName is disabled in freertos_addrinfo, even needed for dns operations.

go2sh, 2021-12-06:
To give some more background:
- prvReadNameField is needed to parse the question and answer sections correctly
- ucName and pcName might be needed with you want to fill the ai_canonname with something independently of the cache

abhidixi11, 2021-12-30:
Hello @go2sh , apologies for late reply. 
Can you  please point me to the code where prvReadNameField is needed but ipconfigUSE_DNS_CACHE or ipconfigDNS_USE_CALLBACKS is not set, same for ucName and pcName. Thanks.

htibosch, 2021-12-30:
Hi @go2sh, thanks for reporting this. One additional question / request: could you attach your copy of FreeRTOSIPConfig.h to your post please?
It should be zipped and you can just drag the file from a file explorer to the edit box.
Thanks

go2sh, 2021-12-31:
I'am back at work on monday. I'll check my code and I can also provide you my workaround.

AniruddhaKanhere, 2022-01-14:
Hello @go2sh, any updates?

go2sh, 2022-01-14:
See the pr #405 for the code I'am using to fix the issue.

go2sh, 2022-01-14:
> Hi @go2sh, thanks for reporting this. One additional question / request: could you attach your copy of FreeRTOSIPConfig.h to your post please? It should be zipped and you can just drag the file from a file explorer to the edit box. Thanks

the required option in the config is just `#define ipconfigUSE_DNS_CACHE 0`

go2sh, 2022-01-14:
In the integration branch it seams to be fixed already.

AniruddhaKanhere, 2022-01-14:
> In the integration branch it seams to be fixed already.

Yes, that branch is not for IPv6. That branch is refactoring IPv4 source code to allow us to provide 100% test coverage and to compartmentalize the codebase.

moninom1, 2023-09-21:
The issue was resolved long back by decoupling DNS cache from the core DNS functionality in the IPv4 main. And now the main and IPv6 branch has merged so issue seems to be resolved in both. Please feel free to reopen the issue if you still feel something is missing. Thank you for the contribution.

closed 2023-09-21 by moninom1