FreeRTOS/FreeRTOS-Plus-TCP #413: ipconfigRAND32() and configRAND32() are still very much in use

fhars, 2022-01-27:
The history file claims they are no longer in use:
https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/b6eac0ca7df8c8935cb840fd9642f6533a0da2e6/include/FreeRTOS_IP.h#L361
but the source code deigns to disagree:
https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/search?q=ipconfigRAND32

There are over a hundred uses of these macros in the released FreeRTOSv202112.00

AniruddhaKanhere, 2022-01-27:
Hello @fhars,

Thank you for bringing this to our attention.

#### For FreeRTOS+TCP repository:
Apologies, but I could not find references of `ipconfigRAND32` in any of the core files (ignoring the FreeRTOSIPConfig.h files) in FreeRTOS+TCP repository. The only place I saw it is in FreeRTOSIPConfigDefaults.h. I `grep`-ed for `configRAND` which should cover ipconfigRAND[32/64].

#### For FreeRTOS repository
I found a lot of instances of ipconfigRAND32 in the demos. Not in any of the core source files. We will fix that soon.

I ignored FreeRTOSIPConfig.h and FreeRTOSConfig.h files since they are meant for configuration and are not part of the 'core' source files.

Thanks,
Aniruddha

fhars, 2022-01-28:
It is all over the tar ball distributed from https://www.freertos.org/a00104.html.

The use is pervasive in the parts where someone who wants to start a new project would begin to look: all over the Demo/ directory and the official documentation: https://www.freertos.org/FreeRTOS-Plus/FreeRTOS_Plus_TCP/TCP_IP_Configuration.html#ipconfigRAND32

htibosch, 2022-01-28:
The macro `ipconfigRAND32()` has existed indeed. It was called in several places, for instance when finding a new "random" port number to bind a socket. And also, to determine the initial sequence number when starting a TCP connection.

FreeRTOS and its libraries has the habit of staying downward compatible. In FreeRTOS_IP.h you will still find code like this:

~~~c
#define xSocket_t                     Socket_t
~~~

The above was an innocent change.

But the function-like macro `ipconfigRAND32()` had a serious problem. I can happen that the random number generator ( RNG ) is broken.  In some systems, `ipconfigRAND32()` returned zero when the RNG was broken. However, the value zero is, of course, a valid random number.

In order to solve this ambiguity, the following applicatiob hook was introduced:

~~~c
/* "xApplicationGetRandomNumber" is declared but never defined, because it may
 * be defined in a user module. */
    BaseType_t xApplicationGetRandomNumber( uint32_t * pulNumber );
~~~

It comes with the following comment:

~~~c
/* This xApplicationGetRandomNumber() will set *pulNumber to a random number,
 * and return pdTRUE. When the random number generator is broken, it will return
 * pdFALSE.
 * The function is defined in 'iot_secure_sockets.c'.
 * If that module is not included in the project, the application must provide an
 * implementation of it.
 * The macro's ipconfigRAND32() and configRAND32() are not in use anymore. */
~~~

I hope that this makes it clear?

fhars, 2022-01-31:
Yes, that is perfectly clear, but still doesn't change anything about the fact that have to rifle through the code of the implementation and the git history to find out about it, since all the examples and all the documentation still look as if the old, broken way is the way to go.

AniruddhaKanhere, 2022-01-31:
@fhars, of course, it is not correct. It misguides people to use incorrect function/macros.
To rectify that - I had created a PR to fix it and it has been merged. If you pull in the latest version of the main branch in FreeRTOS repository, it will have quite less instances of `ipconfigRAND32` calls. I am working on PRs in other repositories which are using this macro. They should be merged soon.

The tar ball that you mentioned has not been updated though - it will only be updated in the next release of FreeRTOS.

AniruddhaKanhere, 2022-02-01:
Since all macros that I found have been removed from the latest version of the code, I am closing this issue.
@fhars thank you for reporting this. If you have any more questions/concerns please feel free to open another issue or reopen this one.

Thanks again,
Aniruddha

closed 2022-02-01 by AniruddhaKanhere