FreeRTOS/FreeRTOS-Plus-TCP #414: FreeRTOS TCP V2.4.0: DHCP Fails

RuStrannik, 2022-01-31:
So, before, up until FreeRTOS TCP V2.3.3 my device used to cold boot (fist-time power-on) with following code:

```
int main(void) {
	bsd_init();
	const uint8_t addr_dummy[4] = { 0, 0, 0, 0 };
	FreeRTOS_IPInit(addr_dummy, addr_dummy, addr_dummy, addr_dummy, bsd_get_MAC_Addr());
	// {"create tasks"}
	vTaskStartScheduler();
	bsd_critical_fail_trap(__func__);
};//main()
```
And everything worked flawlessly, until I tried FreeRTOS TCP V2.4.0, which caused the following problem: the system completely fails to digest received DHCP responce. Inifinitely.

Debugging revealed the following problem:
Before, in `FreeRTOSv202107.00/FreeRTOS-Plus/Source/FreeRTOS-Plus-TCP/FreeRTOS_UDP_IP.c`, Line 318
`vARPRefreshCacheEntry()` was called unconditionally. And everything worked.

Now, in file `FreeRTOSv202112.00/FreeRTOS-Plus/Source/FreeRTOS-Plus-TCP/FreeRTOS_UDP_IP.c`, Line 322
`xCheckRequiresARPResolution()` is called first instead. Inside, it checks if' 'SourceIPAddress' belongs to the same subnet. Obviously, when network is not initialized yet, this check fails, so network never gets initialized.

**The solution is to call `FreeRTOS_IPInit()` with NetMask of `255.255.255.255` on cold (first) boot.**

I would not call it a bug, but rather a significant change in behavior, which was not reflected in release notes and caused quite a bit of time loss.

htibosch, 2022-01-31:
Hello @RuStrannik, I am sorry to hear that you had these problems with +TCP V2.4.0.
And thanks a lot for reporting this. We were not yet aware of this change in behaviour.
I will sort it out and come back to it.
Hein

htibosch, 2022-01-31:
As you see here above, I already created a PR to repair this.
Thanks for sorting out the problem and reporting about it.
Hein

RuStrannik, 2022-01-31:
@htibosch, thank you for prompt response, that solves the problem.

closed 2022-01-31 by RuStrannik

AniruddhaKanhere, 2022-01-31:
Hello @RuStrannik,

Apologies for the trouble the change has caused you. We added the change (using `xCheckRequiresARPResolution`) since that is what RFCs require - we need to send out an ARP Packet for any new local subnet IP we encounter. But as you correctly pointed out, this caused an unforeseen issue with DHCP.

I am reviewing the PR that @htibosch has raised. After minor fixes, we shall merge it soon to the mainline of the repository.

RuStrannik, 2022-01-31:
I agree, having `xCheckRequiresARPResolution()` totally makes sense, I'm glad that this side effect got fixed so quickly.