FreeRTOS/FreeRTOS-Plus-TCP #430: Updating minor and build number

Dweb2, 2022-03-07:
Hi,
I think that the minor (and build) counters need to be incremented here:
https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/55b3b94eb8b4cfae320d20386e8c3f5b1dea13be/include/FreeRTOS_IP.h#L45

AniruddhaKanhere, 2022-03-10:
Hello @Dweb2, thanks for pointing the issue out to us.

I shall fix that ASAP. I will also make note so that it doesn't happen again.

> Maybe write a macro which automatically joins the numbers to a string?

I was thinking something similar to the following?
```
#define STR_HELPER(x) #x
#define STR(x) STR_HELPER(x)

#define ipFR_TCP_VERSION_MAJOR     2
#define ipFR_TCP_VERSION_MINOR     4
#define ipFR_TCP_VERSION_BUILD     0

#define ipFR_TCP_VERSION_NUMBER    "V" STR(ipFR_TCP_VERSION_MAJOR) "." STR(ipFR_TCP_VERSION_MINOR) "." STR(ipFR_TCP_VERSION_BUILD)
```

Dweb2, 2022-03-11:
@AniruddhaKanhere, sounds good!

aggarg, 2022-03-14:
This PR updates version numbers - https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/432

closed 2022-03-16 by aggarg