FreeRTOS/FreeRTOS-Plus-TCP #436: ipconfigTCP_KEEP_ALIVE_INTERVAL

kl-ti, 2022-03-15:
Hi,
i have discovered, that the ipconfigTCP_KEEP_ALIVE_INTERVAL define is missing its name inside the ifndef:
https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/d30e3a7f23d3a4a62678f3b69d98529db25ce20f/include/FreeRTOSIPConfigDefaults.h#L963-L965

AniruddhaKanhere, 2022-03-15:
Hello @kl-ti,

Thanks for taking the time to report this issue.
This bug was added with this PR: https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/391.

I have raised a new PR to fix this: https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/437.

Thanks!

AniruddhaKanhere, 2022-03-15:
Closing this issue now since it was fixed with PR 437.

Thanks again for bringing the issue to our attention @kl-ti.