FreeRTOS/FreeRTOS-Plus-TCP #447: TM4C network interface ignores MAC address passed to FreeRTOS_IPInit

jonathangjertsen, 2022-04-06:
When configuring the Ethernet PHY with a MAC address, TM4C/NetworkInterface.c uses an internal function called `_ethernet_mac_get` which in turn reads the two non-volatile "user registers". This is similar to what TI's example code does, and it works with the evaluation boards which come pre-programmed with a MAC address in these registers. However, the user might want to store the MAC address somewhere else, and then things break.

I believe the correct function to use for this is `FreeRTOS_GetMACAddress`, which returns the MAC address that was configured when calling `FreeRTOS_IPInit`.

Until this is resolved, a possible workaround is to call `MAP_FlashUserSet` before initializing the IP stack.

@jscott-hotstart Do you agree?

htibosch, 2022-04-06:
Normally, the MAC-address of the interface is passed when calling `FreeRTOS_IPInit()`.
And indeed, when any function wants to know the "configured" MAC-address, it should call `FreeRTOS_GetMACAddress()`.

There is also a function `FreeRTOS_UpdateMACAddress()`, which registers a new address. Be aware that that function does not change the MAC-address filter of the EMAC.

In my projects, I often have a 1-wire chip that stores a unique MAC-address. The firmware will read the address before calling `FreeRTOS_IPInit()`.

In the [IPv6/multi branch](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/labs/ipv6_multi), there are multiple interfaces, each with its own MAC-address. The above functions will not be available anymore. In stead, there will be multiple end-points, each one with a MAC- and an IP-address.

jscott-hotstart, 2022-04-06:
@jonathangjertsen, I do agree, but changes coming to +TCP re: IPv6 make the solution a little murky for me.

@htibosch What do you propose that we do in this case? Should we create a PR to solve the issue in the current +TCP branch, then address the issue for the driver in IPv6/multi whenever the TM4C driver is merged there?

htibosch, 2022-04-08:
As you know:

The application calls `FreeRTOS_IPInit()` which will store the IP- and MAC-address in variables.
Later, `xNetworkInterfaceInitialise()` is called, which will retrieve the MAC-address by calling `FreeRTOS_GetMACAddress()`. It will set a MAC filter in `MAP_EMACAddrSet()`.

In simple TCP/IP projects, the MAC-address will never change.

No general provisions were made to update the MAC-address stored in the EMAC. The function `FreeRTOS_UpdateMACAddress()` only updates a variable, not the filter.

What about adding a function called `TM4C_UpdateMACAddress()_`?

I would recommend:

Could you make `_ethernet_mac_get()` public, giving it a clear name like `TM4C_ethernet_mac_get()`? When that is available, the user can decide to use it and pass it to `FreeRTOS_IPInit()`.

I would propose the same approach for the IPv6/multi driver. But let's first get this version done.

jonathangjertsen, 2022-04-08:
As far as I can tell, storing the MAC address in the user registers in the particular format that `_ethernet_mac_get` expects is just a convention from TI's example code. There is nothing in the reference manual about this, for example. However there is support for this convention in their tools such as LMFLasher, so it could be nice to introduce a header file exposing the functionality of `_ethernet_mac_get`. Then the name should indicate that it is reading from the user registers.

I don't think it is important to support updating the MAC address in the hardware.

AniruddhaKanhere, 2022-06-14:
@jonathangjertsen I shall close this issue now - please feel free to open a new issue or reopen this one if you have any questions.

If you want to discuss something - you can create a post on [FreeRTOS-forums](https://forums.freertos.org/).

closed 2022-06-14 by AniruddhaKanhere