FreeRTOS/FreeRTOS-Plus-TCP #449: FreeRTOS+TCP library doesn't compile with static allocation turned on and dynamic allocation turned off

AniruddhaKanhere, 2022-04-07:
The library can use queue and IP-task stack which was allocated statically.

Currently, the stack builds when `configSUPPORT_DYNAMIC_ALLOCATION == 1` regardless of the value of `configSUPPORT_STATIC_ALLOCATION `.

However, with `configSUPPORT_STATIC_ALLOCATION == 1` and `configSUPPORT_DYNAMIC_ALLOCATION == 0`, the stack fails to build since it is using `pvPortMalloc` in myriad socket functions.

pedroluis5, 2025-05-17:
Has anyone found a way to use the FreeRTOS+TCP library with fully static memory allocation?

tony-josi-aws, 2025-05-19:
@pedroluis5 

Thanks for your interest in FreeRTOS+TCP.
We haven't had time to work on this feature yet. We intend to ship this by the end of the year.