FreeRTOS/FreeRTOS-Plus-TCP #454: Compile issue with IAR

renesas-adam-benson, 2022-04-19:
Versions:
- IAR Build Tools for ARM v9.20.2
- FreeRTOS Plus TCP v2.4.0

Hi, when compiling with IAR I am seeing semicolon errors for certain packed structs:

`"FreeRTOS/FreeRTOS-Plus/Source/FreeRTOS-Plus-TCP/include/FreeRTOS_DNS.h",140 Error[Pe1644]: definition at end of file not followed by a semicolon or a declarator`

I am compiling with pack_struct headers included in the repo for IAR (https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/main/portable/Compiler/IAR).

The issue seems to stem from certain structs being moved from source to header files. For example:

```c
/* DNS answer record header. */
#include "pack_struct_start.h"
struct xDNSAnswerRecord
{
    uint16_t usType;       /**< Type of DNS answer record. */
    uint16_t usClass;      /**< Class of DNS answer record. */
    uint32_t ulTTL;        /**< Number of seconds the result can be cached. */
    uint16_t usDataLength; /**< Length of the data field. */
}
#include "pack_struct_end.h"
typedef struct xDNSAnswerRecord DNSAnswerRecord_t;
```

This struct was moved from `FreeRTOS_DNS.c` to `FreeRTOS_DNS.h`. Previous versions of FreeRTOS+TCP before these changes compile fine with IAR.

AniruddhaKanhere, 2022-04-19:
Hello @renesas-adam-benson,

Thank you for reporting this issue to us. We are looking into it.

Aniruddha

AniruddhaKanhere, 2022-04-22:
Hello @renesas-adam-benson,

I tried building a project using FreeRTOS+TCP V2.4.0 with IAR embedded workbench for ARM 9.20.4.47112 (which is the latest) but did not encounter any issues.
The project built successfully.

Can you put your project on GitHub somewhere so that I may clone it and see what the issue is since I cannot replicate your issue locally?

Thanks,
Aniruddha

renesas-adam-benson, 2022-04-22:
Hi @AniruddhaKanhere,

Thanks for the response. I took another look and turns out one of our build configurations had an issue in the order of compilation. I'll close the issue.

closed 2022-04-22 by renesas-adam-benson