FreeRTOS/FreeRTOS-Plus-TCP #456: FreeRTOSIPConfigDefaults define error

holden-zenith, 2022-04-22:
This file is written as if ipconfigEVENT_QUEUE_LENGTH and ipconfigNETWORK_MTU aren't user defined then they will be set to a default. However, the checks 
`#if ( ipconfigEVENT_QUEUE_LENGTH < ( ipconfigNUM_NETWORK_BUFFER_DESCRIPTORS + 5 ) )
    #error The ipconfigEVENT_QUEUE_LENGTH parameter must be at least ipconfigNUM_NETWORK_BUFFER_DESCRIPTORS + 5
#endif

#if ( ipconfigNETWORK_MTU < 46 )
    #error ipconfigNETWORK_MTU must be at least 46.
#endif`
prevent the defaults from ever being used because user definitions are required to pass these checks.

AniruddhaKanhere, 2022-04-25:
Hello @holden-zenith, thank you for bringing this to our attention. I have created PR https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/460 to resolve this issue.

AniruddhaKanhere, 2022-04-25:
I also created a similar PR in IPv6 branch https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/461.

closed 2022-04-25 by holden-zenith