FreeRTOS/FreeRTOS-Plus-TCP #540: [Feature Request] STM32Hxx port: Duplicate MDIO Clock Config setup in HAL

Dweb2, 2022-08-12:
On the current version: ETH_MAC_MDIO_ClkConfig() (https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/013b591124519a91fc81b52209474db6167fe49f/source/portable/NetworkInterface/STM32Hxx/stm32hxx_hal_eth.c#L2741) and HAL_ETH_SetMDIOClockRange() (https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/013b591124519a91fc81b52209474db6167fe49f/source/portable/NetworkInterface/STM32Hxx/stm32hxx_hal_eth.c#L2180) are effectively identical functions which set the management clock. They get called from different contexts.

Maybe consider using just one function.

AniruddhaKanhere, 2022-08-12:
Hello @Dweb2, thank you for bringing this to our attention. We are looking into it and will get back to you soon

shubnil, 2023-02-13:
Hello @Dweb2, We missed this and will be looking into this now.

ActoryOu, 2023-02-16:
Hi @Dweb2,
Thanks for your contribution. We removed duplicate function HAL_ETH_SetMDIOClockRange by #711.
I'm going to close this issue.

Thanks.

closed 2023-02-16 by ActoryOu