FreeRTOS/FreeRTOS-Plus-TCP #549: FreeRTOS+TCP may not be following the TCP standard for connection termination

AmPaschal, 2022-09-08:
**Describe the bug**
I used FreeRTOS to create a server that I connected to a client application.
When I want to terminate the connection, I call close() on my client application. This sends the FIN+ACK frame to FreeRTOS.

According to the TCP standard, FreeRTOS is supposed to first acknowledge the FIN+ACK by sending an ACK, before sending it's own FIN+ACK to close its side of the connection.

But FreeRTOS only sends a FIN+ACK in response to the client's FIN+ACK.

I am not sure if this is allowed by the TCP standard.

**Target**

- Development board: Linux port

**Host**
- Host OS: Linux
- Version: Ubuntu 20.04

**To Reproduce**
- Clone the FreeRTOS+TCP demo application and modify it to run as a TCP server
- Use tshark to monitor the packets to and from FreeRTOS
- Connect to FreeRTOS using a client application.
- Let the client application call the close function after concluding its transactions.
- Monitor the tshark or Wireshark logs.

**Expected behavior**
A concise description of what you expected to happen.

**Screenshots**=
[image]

**Sample Client Application**
[sample_tcp.zip](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/9528756/sample_tcp.zip)

AmPaschal, 2022-09-13:
Hi,

Please, I am adding this comment as a reminder of this issue.

htibosch, 2022-09-13:
Hello @AmPaschal, I have seen both behaviours :
~~~
Party-1 : FIN + ACK
Party-2 : ACK
Party-2 : FIN + ACK
Party-1 : ACK
~~~
as well as:
~~~
Party-1 : FIN + ACK
Party-2 : FIN + ACK
Party-1 : ACK
~~~
I am reading a [Wiki page](https://en.wikipedia.org/wiki/Transmission_Control_Protocol) that says:

> It is also possible to terminate the connection by a 3-way handshake,
> when host A sends a FIN and host B replies with a FIN & ACK
> (combining two steps into one) and host A replies with an ACK.[17]

I think that the behaviour doesn't have any down-sides to it, it is just a bit quicker.

I just observed that my Windows laptop also responded with a 3-way handshake.

[image]

alfred2g, 2022-09-17:
I believe this is a non issue after reading the comment of @htibosch.
Will be closing the issue, please reopen if we missed something.

closed 2022-09-17 by alfred2g