FreeRTOS/FreeRTOS-Plus-TCP #552: [BUG]

phelter, 2022-09-10:
When compiling with: and using linux port, there are missing symbols.

[BufferAllocation_1.c](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/source/portable/BufferManagement/BufferAllocation_2.c)
[linux/NetworkInterface.c](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/source/portable/NetworkInterface/linux/NetworkInterface.c)

The missing symbols are:
```bash
FreeRTOS_Plus_TCP/portable/BufferManagement/BufferAllocation_1.c:189: undefined reference to `vNetworkInterfaceAllocateRAMToBuffers'
```

**Target**
- Development board: linux (for simulation)
- Instruction Set Architecture: x86
- IDE and version: VSCode with custom CMake build script that builds the FreeRTOS similar to the Kernel.
- Toolchain and version: clang13

**Host**
- Host OS: Ubuntu
- Version: 18.04 (Windows WSL-2)

**To Reproduce**
- compile for linux port - and use BufferAllocation_1.c 
- Linking will not occur due to missing symbol `vNetworkInterfaceAllocateRAMToBuffers`

**Expected behavior**
- support for linux port and both buffer allocation schemes.

**Screenshots**
N/A

**Wireshark logs**
N/A

**Additional context**
None.

htibosch, 2022-09-10:
When you put `freertos vNetworkInterfaceAllocateRAMToBuffers` into a search machine, you will get an answer to your question, for instance [here](https://www.freertos.org/FreeRTOS-Plus/FreeRTOS_Plus_TCP/Embedded_Ethernet_Porting.html#vNetworkInterfaceAllocateRAMToBuffers).

Note that the function is only needed when using [BufferAllocation_1.c](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/source/portable/BufferManagement/BufferAllocation_1.c).

phelter, 2022-09-12:
Yes I agree it should be defined, however it isn't in the [linux/NetworkInterface.c](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/source/portable/NetworkInterface/linux/NetworkInterface.c), which means when using BufferAllocation_1.c and `FREERTOS_PORT = GCC_POSIX` (as defined in the FreeRTOS-Kernel) - you get compiler warnings about missing symbols.

The solution is to add a stub for the `vNetworkInterfaceAllocateRAMToBuffers` into the `linux/NetworkInterface.c` so that it can operate with BufferAllocation_1.c.  This is currently missing from this particular port and should be considered the bug.

What I was expecting was to be able to use BufferAllocation_1.c in addition to the linux port (out of the box).  This is currently not the case.

htibosch, 2022-09-12:
There was a reason not to provide `vNetworkInterfaceAllocateRAMToBuffers()` in the distribution: most implementations of that function declare a large static array:
~~~c
uint8_t ucNetworkPackets[ ipconfigNUM_NETWORK_BUFFER_DESCRIPTORS * ETH_MAX_PACKET_SIZE ];
~~~

My compiler uses segments for code and for RAM, so that unused declarations will not waste resources.

We have thought of introducing a config macro like e.g. `ipconfigBUFFER_ALLOCATION_1`. But we didn't want to oblige the user to have this macro correspond with the linker settings. What we did do is:
~~~
const BaseType_t xBufferAllocFixedSize = pdTRUE; /* true in BufferAllocation_1.c
const BaseType_t xBufferAllocFixedSize = pdFALSE; /* false in BufferAllocation_1.c
~~~
So it can be tested in the code. MISRA doesn't like the test involving a const value :-)

paulbartell, 2022-09-12:
@htibosch Do you think it makes sense to heap allocate the network buffers in this case given that this primarily affects the windows / posix simulator ports?

phelter, 2022-09-12:
I understand the reason not having unused resources etc., however the main reason for having the `linux` build is to allow users to natively test their configuration of `FreeRTOS` along with their own code on their native platform (linux) as opposed to on the platform they intend to target (eg. an Arm cortex-a9 in this particular case).

For the configuration portion I implemented that in CMakeLists.txt file similar to the : [FREERTOS_HEAP](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/c09187e73b4d1258f11900a7da163fa3c33b5a3a/CMakeLists.txt#L185) definition to select which of the two buffer allocation techniques to use.

For linux builds I believe it is in `FreeRTOS's` best interest (higher priority) to provide the most fully functional simulation environment rather than worry about wasted resources - i.e. segments of code and RAM for a singular unused declaration when the platform is linux.  If this is not the case, then there are [link-time-optimizations](https://gcc.gnu.org/wiki/LinkTimeOptimization) that should take care of this while also supporting the the removal of unused symbols (when `BufferAllocation_2.c` is used).

Please consider this a feature request then and/or add in documentation that BufferAllocation_1.c is not supported in the simulation ports (linux/windows).

phelter, 2022-09-12:
Another alternative to fix the issue regarding unused symbols is to add an additional file inside the [source/portable/NetworkInterface/linux directory ](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/1ab6eb88857cf1011bf1f8349b43a1c34c7ced0f/source/portable/NetworkInterface/linux).  Eg: `BufferAllocation_1_port.c` which provides an implementation of the `vNetworkInterfaceAllocateRAMToBuffers()` for simulation purposes, and only gets added to the build when [source/portable/BufferManagement/BufferAllocation_1.c](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/1ab6eb88857cf1011bf1f8349b43a1c34c7ced0f/source/portable/BufferManagement/BufferAllocation_1.c) is compiled.

Here's the sample FreeRTOS - portable CMakeLists.txt file feel free to use in your design to provide CMake support to this repo.
```cmake
# FreeRTOS_Plus_TCP/portable/CMakeLists.txt

set(FREERTOS_TCP_BUFFER_ALLOCATION "1" CACHE STRING "FreeRTOS buffer allocation model number. 1 .. 2.")

add_library( freertos_tcp_port STATIC )

target_sources( freertos_tcp_port
  PRIVATE
    BufferManagement/BufferAllocation_${FREERTOS_TCP_BUFFER_ALLOCATION}.c

    $<$<C_COMPILER_ID:GNU>:Compiler/GCC/pack_struct_end.h>
    $<$<C_COMPILER_ID:GNU>:Compiler/GCC/pack_struct_start.h>

    NetworkInterface/Common/phyHandling.c
    NetworkInterface/include/phyHandling.h

    $<$<STREQUAL:${FREERTOS_PORT},GCC_ARM_CA9_ZYNQ>:
      NetworkInterface/Zynq/NetworkInterface.c
      NetworkInterface/Zynq/uncached_memory.c
      NetworkInterface/Zynq/uncached_memory.h
      NetworkInterface/Zynq/x_emacpsif_dma.c
      NetworkInterface/Zynq/x_emacpsif_hw.c
      NetworkInterface/Zynq/x_emacpsif_hw.h
      NetworkInterface/Zynq/x_emacpsif_physpeed.c
      NetworkInterface/Zynq/x_emacpsif.h
      NetworkInterface/Zynq/x_topology.h>

    $<$<STREQUAL:${FREERTOS_PORT},GCC_POSIX>:NetworkInterface/linux/NetworkInterface.c>
    # TODO(phelter): See: https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/issues/552
    $<$<STREQUAL:${FREERTOS_PORT},GCC_POSIX>:NetworkInterface/linux/BufferAllocation_1_port.c>
    $<$<AND:$<STREQUAL:${FREERTOS_PORT},GCC_POSIX>,$<EQUAL:${FREERTOS_TCP_BUFFER_ALLOCATION},1>>:NetworkInterface/linux/BufferAllocation_1_port.c>
)

if( FREERTOS_PORT STREQUAL "GCC_POSIX")
include(FindPCAP.cmake)
endif()

target_include_directories( freertos_tcp_port
  PUBLIC
    include
    $<$<C_COMPILER_ID:GNU>:Compiler/GCC>
    NetworkInterface/include
    # TODO(phelter): Remove this shouldn't be here.
    NetworkInterface/Zynq # for uncached_memory.h in FAT.
  PRIVATE
    # for Zynq/x_emacpsif.h - why is it with Zynq and without elsewhere.
    $<$<STREQUAL:${FREERTOS_PORT},GCC_ARM_CA9_ZYNQ>:NetworkInterface> 
    $<$<STREQUAL:${FREERTOS_PORT},GCC_POSIX>:${PCAP_INCLUDE_DIR}> 

)

# TODO(phelter): Fix these or check fixed in newer version
target_compile_options( freertos_tcp_port
  PRIVATE
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-cast-align>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-conditional-uninitialized>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-documentation>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-empty-translation-unit>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-extra-semi-stmt>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-gnu-statement-expression>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-implicit-int-conversion>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-missing-noreturn>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-missing-prototypes>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-pointer-to-int-cast>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-newline-eof>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-padded>
[... 94 lines cut ...]
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-macro-redefined>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-missing-noreturn>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-missing-variable-declarations>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-packed>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-padded>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-reserved-identifier>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-reserved-macro-identifier>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-shorten-64-to-32>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-sign-conversion>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-tautological-constant-out-of-range-compare>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-undef>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-unused-but-set-variable>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-unused-function>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-unused-macros>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-unused-variable>
)

#TODO(phelter): should separate out portable portion into separate lib.

target_link_libraries( freertos_tcp
  PRIVATE
    freertos_kernel
    freertos_tcp_port
)
```

paulbartell, 2022-09-12:
@phelter I think the intention of the different buffer allocation modes it to allow the end-developer to choose their preferred implementation and potentially not use any of the provided buffer allocation schemes if they have a different preference.

I'm hesitant to use your CMakeLists.txt as-is since there is not a one-to-one correlation between FreeRTOS port and FreeRTOS+TCP port, especially for ARM Cortex-M devices. It certainly brings up a useful discussion on what a CMakeLists.txt for FreeRTOS+TCP should look like.

phelter, 2022-09-12:
@paulbartell - Yes that's fine.  It was never my intention on suggesting the CMakeLists.txt should be used verbatim.  I thought I'd provide it so you had a reference to what I was compiling and for what reason.  The Xilinx port for their 7000 Zynq components uses `BufferAllocation_1.c` for their implementation.

paulbartell, 2022-09-12:
@phelter : I pushed PR #553  which should address your issued with BufferAllocation1.c and the posix simulator. Feel free to comment on that PR if you have any concerns.

htibosch, 2022-09-14:
> @htibosch Do you think it makes sense to heap allocate the network buffers in this case given that this primarily affects the windows / posix simulator ports?

Yes I think that would make sense.

johnrhen, 2022-09-23:
I'm closing this issue, now that PR #553 has been merged. Please reopen if you have any additional concerns.

closed 2022-09-23 by johnrhen

phelter, 2022-09-23:
I'll add a separate Issue for the changes associated with adding a CMake support.