FreeRTOS/FreeRTOS-Plus-TCP #558: [BUG] Fix Clang/GCC compiler warnings.

phelter, 2022-09-25:
**Describe the bug**
Cleanup the Clang and GNU compiler warnings when `-Wall`, `-Weverything`, and `-Wpedantic` are used.

Start using clang for checking doxygen and other C based linting violations - can use in gitub-actions.

**Target**
- Development board: Linux
- Instruction Set Architecture: Linux
- IDE and version: Visual Code 1.71.1
- Toolchain and version: Clang-13 and GCC 7.5.0

**Host**
- Host OS: Linux
- Version: Ubuntu 18.04

**To Reproduce**
- Based on - https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/557
- Run on Ubuntu 18.04 with Clang 14 and Remove the `target_compile_options(...` could observe failures when it compiles

**Expected behavior**
Code cleaned up and a CI call to Clang and GCC to ensure code quality is adhered to.

**Screenshots**
N/A

**Wireshark logs**
N/A

**Additional context**
The primary reason for doing this is to use Clang and GCC to provide better and more descriptive code.

kstribrnAmzn, 2023-02-15:
Dumping the warnings I found using the QEMU emulator demo compiled with arm-none-eabi-gcc. This is lazy and I'll need to check further, this is just the quickest way I could get to producing warnings.

Setup:
* Mac Pro, I7, Ventura 13.2
* arm-none-eabi-gcc (Arm GNU Toolchain 11.3.Rel1) 11.3.1 20220712
* Flags used - `-Wall -Wextra -Wpedantic`
```
startup.c: In function 'prvGetRegistersFromStack':
startup.c:83:23: error: variable 'psr' set but not used [-Werror=unused-but-set-variable]
   83 |     volatile uint32_t psr; /* Program status register. */
      |                       ^~~
startup.c:82:23: error: variable 'pc' set but not used [-Werror=unused-but-set-variable]
   82 |     volatile uint32_t pc;  /* Program counter. */
      |                       ^~
startup.c:81:23: error: variable 'lr' set but not used [-Werror=unused-but-set-variable]
   81 |     volatile uint32_t lr;  /* Link register. */
      |                       ^~
startup.c:80:23: error: variable 'r12' set but not used [-Werror=unused-but-set-variable]
   80 |     volatile uint32_t r12;
      |                       ^~~
startup.c:79:23: error: variable 'r3' set but not used [-Werror=unused-but-set-variable]
   79 |     volatile uint32_t r3;
      |                       ^~
startup.c:78:23: error: variable 'r2' set but not used [-Werror=unused-but-set-variable]
   78 |     volatile uint32_t r2;
      |                       ^~
startup.c:77:23: error: variable 'r1' set but not used [-Werror=unused-but-set-variable]
   77 |     volatile uint32_t r1;
      |                       ^~
startup.c:76:23: error: variable 'r0' set but not used [-Werror=unused-but-set-variable]
   76 |     volatile uint32_t r0;
      |                       ^~
startup.c: At top level:
startup.c:164:5: error: ISO C forbids conversion of function pointer to object pointer type [-Werror=pedantic]
  164 |     ( uint32_t * ) &Reset_Handler,       /* Reset                -15  */
      |     ^
startup.c:165:5: error: ISO C forbids conversion of function pointer to object pointer type [-Werror=pedantic]
  165 |     ( uint32_t * ) &Default_Handler,     /* NMI_Handler          -14  */
      |     ^
startup.c:166:5: error: ISO C forbids conversion of function pointer to object pointer type [-Werror=pedantic]
  166 |     ( uint32_t * ) &Default_Handler2,    /* HardFault_Handler    -13  */
      |     ^
startup.c:167:5: error: ISO C forbids conversion of function pointer to object pointer type [-Werror=pedantic]
  167 |     ( uint32_t * ) &Default_Handler3,    /* MemManage_Handler    -12  */
      |     ^
startup.c:168:5: error: ISO C forbids conversion of function pointer to object pointer type [-Werror=pedantic]
  168 |     ( uint32_t * ) &Default_Handler4,    /* BusFault_Handler     -11  */
      |     ^
startup.c:169:5: error: ISO C forbids conversion of function pointer to object pointer type [-Werror=pedantic]
  169 |     ( uint32_t * ) &Default_Handler5,    /* UsageFault_Handler   -10  */
      |     ^
startup.c:174:5: error: ISO C forbids conversion of function pointer to object pointer type [-Werror=pedantic]
  174 |     ( uint32_t * ) &vPortSVCHandler,     /* SVC_Handler          -5   */
      |     ^
startup.c:175:5: error: ISO C forbids conversion of function pointer to object pointer type [-Werror=pedantic]
  175 |     ( uint32_t * ) &Default_Handler6,    /* DebugMon_Handler     -4   */
      |     ^
startup.c:177:5: error: ISO C forbids conversion of function pointer to object pointer type [-Werror=pedantic]
  177 |     ( uint32_t * ) &xPortPendSVHandler,  /* PendSV handler       -2   */
      |     ^
startup.c:178:5: error: ISO C forbids conversion of function pointer to object pointer type [-Werror=pedantic]
  178 |     ( uint32_t * ) &xPortSysTickHandler, /* SysTick_Handler      -1   */
      |     ^
startup.c:192:5: error: ISO C forbids conversion of function pointer to object pointer type [-Werror=pedantic]
  192 |     ( uint32_t * ) &EthernetISR,         /* Ethernet              13  */
      |     ^
startup.c: In function 'exit':
startup.c:202:43: error: unused parameter 'status' [-Werror=unused-parameter]
  202 | __attribute__( ( naked ) ) void exit( int status )
      |                                       ~~~~^~~~~~
startup.c: At top level:
startup.c:117:13: error: 'HardFault_Handler' declared 'static' but never defined [-Werror=unused-function]
  117 | static void HardFault_Handler( void ) __attribute__( ( naked ) );
```

kstribrnAmzn, 2023-02-16:
Going to dump what I was able to my setup information and what warnings I was able to find here.

Setup:
* Mac Pro, i7, Ventura 13.2
* Clang version (clang-1400.0.29.202), x86_64-apple-darwin22.3.0

I was able to add the warning flags you requested (`-Wall -Weverything -Wpedantic`) to the default config used in the CI. I built this and saw many warnings but was able to boil them down to a couple things...
1. I was pretty lazy in adding the flags. The warning flags were also applying to the FreeRTOS/Kernel files when the +TCP library was being built. I overcame this by refining the CMake file I had created to ONLY care about the +TCP source files.
2. The #pragma packing push/pop directives were causing the bulk of the warnings I would see. For me, the errors would look something like...
```
/workspace/forks/FreeRTOS-Plus-TCP/source/include/FreeRTOS_DNS_Globals.h:134:14: warning: the current #pragma pack alignment value is modified in the included file [-Wpragma-pack]
    #include "pack_struct_start.h"
             ^
/workspace/forks/FreeRTOS-Plus-TCP/source/portable/Compiler/MSVC/pack_struct_start.h:35:9: note: previous '#pragma pack' directive that modifies alignment is here
#pragma pack( push, 1 )
        ^
In file included from /Users/kstribrn/workspace/forks/FreeRTOS-Plus-TCP/test/build-combination/Common/main.c:38:
In file included from /Users/kstribrn/workspace/forks/FreeRTOS-Plus-TCP/source/include/FreeRTOS_IP.h:448:
In file included from /Users/kstribrn/workspace/forks/FreeRTOS-Plus-TCP/source/include/FreeRTOS_IP_Utils.h:62:
In file included from /Users/kstribrn/workspace/forks/FreeRTOS-Plus-TCP/source/include/FreeRTOS_DNS.h:38:
```

I think the only way to overcome this is to suppress the `-Wpramga-pack` warning. I did this by adding the following lines to the pack_struct_start.h file
```
#pragma clang diagnostic push
#pragma clang diagnostic ignored "-Wpragma-pack"
```

_______
I suspect my strategy here is a little too loose as when I now build, I'm not seeing any warnings. I'll chat with others tomorrow to get some ideas for how this can be refined.

kstribrnAmzn, 2023-02-16:
@phelter Can you elaborate on the Clang/GCC compiler warnings you were seeing? Were they all related to the packing directives? Or were there others?

phelter, 2023-03-06:
@kstribrnAmzn - 

They were:
- improper use of doxygen, missing or wrong parameters
- declaration after statement  - this is a violation of the C90/C89 ISO. which is what FreeRTOS is targeting for their repos.
- Other various issues - please see: https://github.com/phelter/FreeRTOS-Plus-TCP/tree/feature/fixing-clang-gnu-compiler-warnings for a full change.  These do not fix all of them but it's a good start.

kstribrnAmzn, 2023-03-07:
Thanks @phelter! I'll take a look at what you've done, pull what I can in, and come up with a PR. I'll probably break this into several PRs so that the fixes flow faster.

kstribrnAmzn, 2023-03-15:
GCC warnings fixes - https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/798

kstribrnAmzn, 2023-03-16:
The Clang fixes are a little trickier here as there are more false positives that aren't as easily resolved.

Example 1:
```
wait_for_event.h:30:9: error: macro name is a reserved identifier [-Werror,-Wreserved-macro-identifier]
#define _WAIT_FOR_EVENT_H_
```
From [this file](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/ThirdParty/GCC/Posix/utils/wait_for_event.h#L30) in the Kernel. This is because Clang (and other compilers) reserve macros starting with an underscore and capitalized. I have not found a [Clang diagnostic flag](https://clang.llvm.org/docs/DiagnosticsReference.html) for disabling the reserved macro warning.

phelter, 2023-03-16:
@kstribrnAmzn those `false positives` are easily resolved by search and replacing each without the prefix of the `_` (preferred change).

eg:
```c
#ifndef WAIT_FOR_EVENT_H
#define WAIT_FOR_EVENT_H
```

Clang diagnostic flag for disabling this is like all other clang diagnostics by prefixing the diagnostic with  `-Wno-`.

So for this scenario it would be `-Wno-reserved-macro-identifier`

kstribrnAmzn, 2023-03-23:
Thanks for the advice @phelter. I actually found some things couldn't be disabled with the `-Wno-` prefix. IIRC the reserved macro identifier was them which surprised me.

Anyway I have posted a [Clang warnings fix](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/809) PR and my [GCC warnings fix](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/798) has been approved. When you look at these you're probably going to wonder why I left out the _pedantic_ warnings. This was a conscious decision by the team. Pedantic was removed for a couple reasons: other solutions decreased readability and would have caused customer pain either in consuming our library or debugging application problems with it.

If you're happy, then there is no further action on your part. When the Clang PR is merged, I'll close out this issue.

closed 2023-03-24 by kstribrnAmzn