FreeRTOS/FreeRTOS-Plus-TCP #564: Any plans to support a second DNS server?

quartzjer, 2022-10-18:
https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/e0ce54235881ef9f479dd87ecf832642e96a35b7/source/FreeRTOS_DHCP.c#L897

Have hit this multiple times where resolution would fail because the local DHCP's primary DNS was down but secondary was available. Trying to find a quick workaround but it isn't looking like an easy patch.

archigup, 2022-10-19:
Hi, thanks for bringing this up! I'll bring this request to our product team's attention.
If you get it working, we would welcome a PR.

htibosch, 2022-10-19:
Yes it is a good idea. The [ipv6_multi branch](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/labs/ipv6_multi) already supports multiple DNS addresses, and the main (IPv4/single) branch should do so too.

The algorithm is simple: as long as a DNS server replies, stick with it. As soon as a server times out, the next DNS address will be tried. When the last address has been tried, the first entry will be used again.

~~~diff
 typedef struct xNetworkAddressingParameters
 {
     uint32_t ulDefaultIPAddress; /**< The default IP address */
     uint32_t ulNetMask;          /**< The netmask */
     uint32_t ulGatewayAddress;   /**< The gateway address */
-    uint32_t ulDNSServerAddress; /**< The DNS server address */
+    uint32_t ulDNSServerAddress[ ipconfigDNS_ADDRESS_COUNT ]; /**< The DNS server addresses */
     uint32_t ulBroadcastAddress; /**< The Broadcast address */
 } NetworkAddressingParameters_t;
~~~

We will come back to it.

n9wxu, 2022-11-04:
We would love this feature added but we are hard at work on other things so if you are interested in adding the feature and submitting a PR, we would welcome the opportunity to review it.

@htibosch has outlined the algorithm above and there is an example in the ipv6_multi branch.

This issue will be left open until the feature is added.

moninom1, 2023-06-05:
HI @quartzjer

We have added the said feature in the new[ dev/IPv6_integration](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/dev/IPv6_integration)  branch. This is where multiple DNS addresses are handled [ulDNSServerAddresses](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/dev/IPv6_integration/source/include/FreeRTOS_Routing.h#L93) 

Please let us know if you have any other query. If not we will go ahead and close the issue.

Thank you.

moninom1, 2023-06-07:
Closing the issue for now. Let us know if any other information is needed. Thank You

closed 2023-06-07 by moninom1