FreeRTOS/FreeRTOS-Plus-TCP #565: [Feature Request] TCP_NODELAY

quartzjer, 2022-10-18:
Any plans to support a `FREERTOS_SO_NODELAY` or equivalent for low latency streaming applications?

archigup, 2022-10-19:
Hi, NODELAY in other stacks is used to disable Nagle's algorithm; FreeRTOS+TCP does not use Nagle's algorithm and should not delay delivery.

Are you having latency issues with using the stack?

htibosch, 2022-10-19:
The TCP driver calculates the Smoothed Round Trip Time (SRTT), and in case a peer doesn't respond, the RTT indicates the time before a retransmission shall be done.

An exception on these waiting times is the ["fast retransmission"](https://wiki.wireshark.org/TCP_Analyze_Sequence_Numbers).

FreeRTOS+TCP is suitable to play audio, even if there are occasional retransmissions.

~~~
/* Smoothed Round Trip Time, it may increment quickly and it decrements slower */
int32_t lSRTT;
~~~

The waiting time will be doubled after every timeout: `1 x lSRTT`, `2 x lSRTT`, `4 x lSRTT`.

There is a minimum for `lSRTT`, defined as:
~~~
    #define winSRTT_CAP_mS   ( ipconfigTCP_SRTT_MINIMUM_VALUE_MS ) /* Cap in milliseconds. */
~~~

About a year ago, we changed the above define in [PR #387](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/387) because the default was too low ( 50 ms ).

@archigup wrote:

> NODELAY in other stacks is used to disable Nagle's algorithm; FreeRTOS+TCP does not use Nagle's algorithm and should not delay delivery.

Agreed, Nagle was never implemented.

@quartzjer : when you also develop the remove TCP application, make sure that it also disables Nagle by setting the `TCP_NODELAY` option.

> Are you having latency issues with using the stack?

I am curious too. Is it possible to show the audio stream, recorded by e.g. WireSkark? If you want you can create a new subject on the [FreeRTOS forum](https://forums.freertos.org) and post it there.

Thanks

shubnil, 2023-06-14:
Closing this issue and a new ticket can be created if any low latency specific enhancement is needed.

closed 2023-06-14 by shubnil