FreeRTOS/FreeRTOS-Plus-TCP #680: Default IP and does not get updated within FreeRTOS_SetAddressConfiguration()

evpopov, 2023-01-23:
This relates to the "main" branch

If I understand the intent of the developers correctly, then when changing the IP settings, I'm supposed to call FreeRTOS_SetAddressConfiguration() however it only updates  *ipLOCAL_IP_ADDRESS_POINTER and not xNetworkAddressing.ulDefaultIPAddress
The only use for xNetworkAddressing.ulDefaultIPAddress that I found was for DHCP when it falls back to using the default IP.

Here is why I care about xNetworkAddressing.ulDefaultIPAddress

Imagine that somehow a bad combination of firmware, settings, etc finds their way on my FreeRTOS+TCP device and sometimes during the boot process, the device resets, hangs, dies, or simply becomes inaccessible over the network. Call it "bricked". 
I want to prevent this my pausing the boot process, waiting for a "magic" packet and if received, booting into a minimal-but-known-good firmware. To achieve this, I bring up the TCP stack as early during the boot process as possible with as little other initialization as possible. No filesystem, no nothing. This allows me to have confidence that I can receive that packet even in cases where later parts of the boot process kills the device. Later on, during a normal boot sequence, when the magic packet is not received, I init the file system, read my network settings and need to update the TCP stack which has already been initialized. And there in lies the rub.... I can't set xNetworkAddressing.ulDefaultIPAddress which means DHCP will fall back to whatever hardcoded IP I used when I called FreeRTOS_IPInit() instead of whatever default IP I found in my settings file after FreeRTOS_IPInit() was initially called.

My current bandaid fix is to declare an extern to xNetworkAddressing and manually update the structure. That is not a great solution as clearly xNetworkAddressing was not intended to be touched by pesky users like myself.

My proposed fix is as simple as adding
```
xNetworkAddressing.ulDefaultIPAddress = *pulIPAddress;
xDefaultAddressing.ulBroadcastAddress = ( xDefaultAddressing.ulDefaultIPAddress & xDefaultAddressing.ulNetMask ) | ~xDefaultAddressing.ulNetMask;

```
to FreeRTOS_SetAddressConfiguration()

p.s.  Please verify the need for updating xNetworkAddressing.ulBroadcastAddress. I think it never gets updated when DHCP falls back to the default IP but at the same time, calling FreeRTOS_SetAddressConfiguration() might mess up the current DHCP lease if it's present... I haven't looked too deep into that because in my case, after I overwrite xNetworkAddressing, I call FreeRTOS_NetworkDown() just to let everything initialize cleanly.

Thanks for looking into this.

archigup, 2023-01-24:
Hi, thanks for bringing this up; we're looking into it.

AniruddhaKanhere, 2023-02-15:
Hello @evpopov,

Apologies for the late response, I was under the weather.

Have you seen this post and does the workaround mentioned in the post work for you? https://forums.freertos.org/t/freertos-setaddressconfiguration-question/14027

Also, I think that the `xNetworkAddressing` structure should be modified even in APIs very carefully. Because there can be a race condition in case the FreeRTOS+TCP stack is updating the variable while your API is also updating it.

Let me know if that helps.

evpopov, 2023-02-22:
Hi @AniruddhaKanhere,
What I have been doing is exactly the same as what is described in the post you quoted. I simply set all the fields in xDefaultAddressing and then call FreeRTOS_NetworkDown().
... so I guess the quick answer is YES, the workaround works for me. I just find it weird that I have to modify a struct that is defined as an extern in **FreeRTOS_IP_Private.h**. This workaround feels like going around the well-documented API and what I was suggesting with this issue is that the API be fixed so we don't have to resort to setting fields in a struct.

I can see the so called race condition that you are referring to, but it also looks to me that the same race condition exists 
with
xNetworkAddressing.ulNetMask
xNetworkAddressing.ulGatewayAddress
xNetworkAddressing.ulDNSServerAddress
because all of them get set in FreeRTOS_SetAddressConfiguration()
Maybe  FreeRTOS_SetAddressConfiguration() needs a pair of vTaskSuspendAll / xTaskResumeAll

If you feel that this issue is not worth fixing or is not an issue at all, please feel free to close it. I'm sure there are much more important things to do.
Thanks for looking into this.

moninom1, 2023-04-12:
Hi @evpopov 
Thank you for bringing this up.
I would like to let you know that we are working on [dev/IPv6_Integration](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/dev/IPv6_integration) branch, which have IPv6 and IPv4 integrated in one branch and will be merged to main in some time.
In that branch we have moved away from [FreeRTOS_SetAddressConfiguration](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/dev/IPv6_integration/GettingStarted_4.0.0-rc1.md) and we are using FreeRTOS_SetEndPointConfiguration. Which will be updating the global endpoints instead of ipLOCAL_IP_ADDRESS_POINTER. And the same information will be passed to DHCP as well.
So according to what you have mentioned, I think the workaround will not be needed in the new branch.

Let me know if you have any more queries, i will be happy to help.

Thank you

evpopov, 2023-04-13:
Since FreeRTOS_SetAddressConfiguration() is being phased out and because there is a workaround for the original issue I posted, I consider this a moot point and am closing the issue.

Thank you all