FreeRTOS/FreeRTOS-Plus-TCP #773: [BUG] Build Error in FreeRTOS_ND.c

holden-zenith, 2023-03-09:
I am getting xIPv6_GetIPType not defined even when ipconfigUSE_IPv6.
https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/0b67e4551f31529a41e7c3f37a5c281bb1db7da6/source/FreeRTOS_ND.c#L205

tony-josi-aws, 2023-03-10:
Hey @holden-zenith,

Thanks for reporting this. 

Could you please confirm if you are using the latest commit of the [dev/IPv6_integration](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/dev/IPv6_integration) branch. I believe [this PR](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/759) fixes the issue.

moninom1, 2023-03-11:
Hey @holden-zenith 
Is the issue still seen on the latest [dev/IPv6_integration](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/dev/IPv6_integration) branch

htibosch, 2023-03-11:
Please check if you see the declaration in FreeRTOS_Routing.[ch].

> I believe [this PR](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/759) fixes the issue.

The function `xIPv6_GetIPType()` is relatively new. The [dev/IPv6_integration](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/dev/IPv6_integration) branch is still work in progress. But we're glad when users like you become "early adopters" and give feedback. Thank you for that.

closed 2023-03-11 by holden-zenith