FreeRTOS/FreeRTOS-Plus-TCP #807: Request for GCC Makefile for Multi_interface_demo

xeroxx, 2023-03-23:
Hi

Can you provide me with a Makefile for Multi_interface_demo for GCC?
I'm trying to compile this demo for a Cortex-M3 ARM processor.

Thanks,
Saeed

htibosch, 2023-03-23:
Hello Saeed, I'm not sure which "Multi_interface_demo" you are referring to?

We have these branches:
[ipv6_multi](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/labs/ipv6_multi) deprecated
[IPv6_integration](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/dev/IPv6_integration)

The former branch stops developing, but it does have a [demos](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/labs/ipv6_multi/demos) directory.

The second branch is still under development. But things go fast now, many people are involved to make this the new stable "main branch". It will support IPv6, and multiple interfaces and endpoints.

If you want I can give you a copy of the WinSim demo that I use for integration testing IPv6_integration.

Maybe other people can point out where to find makefile projects for this purpose.

tony-josi-aws, 2023-03-23:
Hey @xeroxx ,

We dont have a QEMU Makefile demo yet for the multi interface. But we do have [FreeRTOS_Plus_TCP_Echo_Qemu_mps2](https://github.com/FreeRTOS/FreeRTOS/tree/main/FreeRTOS-Plus/Demo/FreeRTOS_Plus_TCP_Echo_Qemu_mps2) project that uses Makefile, but that uses single interface/endpoint version of FreeRTOS+TCP. As htibosch suggested you could refer the multi interface demo and update the FreeRTOS_Plus_TCP_Echo_Qemu_mps2 to use multi interfaces for your case.

xeroxx, 2023-03-23:
Hello Hein,
Thanks for your reply!

> I'm not sure which "Multi_interface_demo" you are referring to?

I was referring to [ipv6_multi](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/labs/ipv6_multi) which has a demo for Windows simulator.

> If you want I can give you a copy of the WinSim demo that I use for integration testing IPv6_integration.

I checked the demos directory of [IPv6_integration](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/dev/IPv6_integration) and the folder was empty. Can you add the demo to the repository so I can use it to build it for QEMU system simulator?

xeroxx, 2023-03-23:
Thanks for your reply!

> As htibosch suggested you could refer the multi interface demo and update the FreeRTOS_Plus_TCP_Echo_Qemu_mps2 to use multi interfaces for your case.

I'm thinking of doing it in FreeRTOS-Plus/Source/FreeRTOS-Plus-TCP/demos directory. Does it follow the directory structure then? (just in case if I share the final demo with the community)

chinglee-iot, 2023-03-29:
We put the demos in the in this path [FreeRTOS-Plus/Demo](https://github.com/FreeRTOS/FreeRTOS/tree/main/FreeRTOS-Plus/Demo). It is suggested to put the demo in this folder. Developers who are already familiar with the folder structure can find the demo easily.

chinglee-iot, 2023-03-29:
Close this issue first. Feel free to reopen it if you have further question about the multi_interface_demo.

closed 2023-03-29 by chinglee-iot