FreeRTOS/FreeRTOS-Plus-TCP #808: [Feature Request] Driver for Pico W Network Interface (CYW43)

adam-scislowicz, 2023-03-23:
**Is your feature request related to a problem? Please describe.**
Pico W wifi networking is not available for FreeRTOS-Plus-TCP

**Describe the solution you'd like**
Integration of the CYP43 driver as included in the Pico SDK (https://github.com/raspberrypi/pico-sdk).

**Describe alternatives you've considered**
FreeRTOS w/ lwIP: https://mcuoneclipse.com/2023/03/19/ble-with-wifi-and-freertos-on-raspberry-pi-pico-w/

**Additional context**
none

johnrhen, 2023-04-06:
Thanks for submitting this, Adam, we'll add this feature to our input queue.

ActoryOu, 2025-06-06:
Hi @bergp1,
Thanks for your feedback! Sadly, we currently don't have a definitive timeline for its implementation.

BTW, we welcome community contributions! If you're interested in helping develop this feature, please feel free to submit a pull request. Your expertise and input would be greatly appreciated.

Thank you!