FreeRTOS/FreeRTOS-Plus-TCP #846: dev/ipv6_integration pxUDPPayloadBuffer_to_NetworkBuffer() doesn't seem to handle IPv6 at all

evpopov, 2023-04-25:
When called, pxUDPPayloadBuffer_to_NetworkBuffer() blindly subtracts ipUDP_PAYLOAD_IP_TYPE_OFFSET from the buffer pointer

It should subtract either ipUDP_PAYLOAD_OFFSET_IPv4 or ipUDP_PAYLOAD_OFFSET_IPv6 but I can't figure out how to do that. The information is simply not there.

Am I missing something? This is a huge bug and for the past few days I've been trying to figure out what I'm doing wrong, but I don't think it's me anymore. I made a simple patch where I pass the IP type as parameter to pxUDPPayloadBuffer_to_NetworkBuffer() and all of a sudden, LLDP over v6 finally started working.

evpopov, 2023-04-27:
Hi @htibosch,
Your last post helped me realize my mistakes and I found the problem. It is NOT a bug in FreeRTOS+TCP. 
First of all, the reason I was looking into this was because when receiving IPv6 LLMNR frames, my device would hit the assert in `pxUDPPayloadBuffer_to_NetworkBuffer()` and freeze. From the very beginning, I though the issue was with how IPv6 was handled.

The problem comes from a combination of 2 thing:

- My device uses a managed 3 port switch for a physical and that managed switch is configured to add a special VLAN tag to all frames going to the CPU port. The special VLAN tag holds information for the ingress port of the frames. The same VLAN tag is used to define the egress port for outgoing frames and is stripped off before the frames leave the switch to the outside world. This is very useful and required for protocols like LLDP and others. In order to accommodate those special VLAN tags, I have a patch for the TCP stack that adds support for VLAN tagged frames. That patch was very carefully crafted to be transparent to the stack and updates a few definitions so that FreeRTOS+TCP is aware of the tag but simply does not care about them.
- There is some sorcery going on with ipIP_TYPE_OFFSET and I must admit I did not fully understand why it was done. The original definition was that ipIP_TYPE_OFFSET is 6.... Some magic number that I did not understand.... why 6? why not 1...

Here's what caused my confusion:
```
#define ipIP_TYPE_OFFSET            ( 6U )
#define ipUDP_PAYLOAD_IP_TYPE_OFFSET    ( sizeof( UDPPacket_t ) + ipIP_TYPE_OFFSET )
```
At this point, `ipIP_TYPE_OFFSET` to me is a magic number that somehow just works, but there is no information on why it needs to be 6 exactly.

... from FreeRTOS_IP.c
```
pxNetworkBuffer->pucEthernetBuffer[ 0 - ( BaseType_t ) ipIP_TYPE_OFFSET ] = pxIPHeader_IPv6->ucVersionTrafficClass;
...
pxNetworkBuffer->pucEthernetBuffer[ 0 - ( BaseType_t ) ipIP_TYPE_OFFSET ] = pxIPHeader->ucVersionHeaderLength;
```
At this point, I believed that BECAUSE the IP version byte is stored at `pxNetworkBuffer->pucEthernetBuffer[-6]` for BOTH IPv4 and IPv6 than it must be USED for both  IPv4 and IPv6 later on. I was wrong! The byte at `pxNetworkBuffer->pucEthernetBuffer[-6]` is ONLY used when dealing with IPv4 frames.
With this erroneous idea in mind, and with the knowledge that IPv6 headers are longer than IPv4 headers, I was absolutely certain that the distance from the UDP payload to `pxNetworkBuffer->pucEthernetBuffer[-6]` is different for IPv4 and IPv6 frames and therefor `pxUDPPayloadBuffer_to_NetworkBuffer()` must subtract different number from the payload header when dealing with the two IP versions.

What I kept failing to realize is that storing the IPv4 version byte at the magic location pxNetworkBuffer->pucEthernetBuffer[-6] made an IPv4 frame appear as an IPv6 frame as far as pxUDPPayloadBuffer_to_NetworkBuffer() is concerned.

That little trick allows `pxUDPPayloadBuffer_to_NetworkBuffer()` to subtract the same constant from a payload buffer and end up at a byte that tells it if this is an IPv4 or IPv6 frame. The function can then subtract a variable amount to get a NetworkBufferDescriptor_t pointer.

Here's the problem though......
My VLAN tag aware implementation modifies `EthernetHeader_t` to add the special VLAN tag. This along with some other definition modifications allow FreeRTOS+TCP to operate just fine in the presence of these VLAN tags. However adding 4 bytes to the Ethernet header changes the math. Since ipIP_TYPE_OFFSET is a hardcoded constant, the result is that the IP version byte of all frames still gets stored at `pxNetworkBuffer->pucEthernetBuffer[-6]` however that location is no longer the same distance from the UDP payload as the distance in an IPv6 frame. Here's why:
The distance between the UDP payload and the version byte of an IPv6 frame is always 40+8=48 bytes.
For an IPv4 frame, the original calculation was: ` sizeof(EthernetHeader_t) + sizeof(IPHeader_t) + sizeof(UDPHeader_t) + 6` and that used to be 48 when EthernetHeader_t was the original 14 bytes long.
In my implementation though, EthernetHeader_t became 18 bytes, causing ipUDP_PAYLOAD_IP_TYPE_OFFSET to automatically become 18+20+8+6 = 52. This new 52 number is actually correct for an IPv4 frame and so my TCP stack worked just fine with IPv4 frames and the special VLAN tags. HOWEVER, that same number was used for IPv6 frames as well causing `pucIPType` in `pxUDPPayloadBuffer_to_NetworkBuffer()` to point at the wrong byte and consequently the assert to fail. 

Now that I understand all that and there is no more "sorcery" surrounding ipIP_TYPE_OFFSET the solution that I'd like to propose is very simple:
Change
```
#define ipIP_TYPE_OFFSET            ( 6U )
...
#define ipUDP_PAYLOAD_IP_TYPE_OFFSET    ( sizeof( UDPPacket_t ) + ipIP_TYPE_OFFSET )
```
to
```
#define ipUDP_PAYLOAD_IP_TYPE_OFFSET    ( sizeof( UDPHeader_t ) + sizeof( IPHeader_IPv6_t ) )
...
#define ipIP_TYPE_OFFSET    ( sizeof( UDPHeader_t ) + sizeof( IPHeader_IPv6_t ) - sizeof( UDPPacket_t ) )
```
Those changes fix my VLAN tag aware implementation because the above automatically calculates `ipIP_TYPE_OFFSET` to be 2 which is the correct magic number for my implementation.

Here's my reasoning:
- `pxUDPPayloadBuffer_to_NetworkBuffer()` has to be able to treat an IPv4 frame the same as an IPv6 frame. The leading factor here is the IPv6 frame's distance between the UDP payload and the location of the IP versio [... cut ...]
- Those changes take the guesswork out of the equation and prevent people like me ( that know enough to be dangerous ) from shooting themselves in the foot like I did.

In closing, I'd like to keep this issue open so we can finalize the discussion and then I'd like to do a PR that incorporates the changes I mentioned + 1-2 more minor alterations that aim at preventing someone else from going down that same path that I went down. I hope it all makes sense now @htibosch and thanks a lot for helping me with this issue!

shubnil, 2023-04-28:
Thanks @evpopov and @htibosch . Let us close the issue.

closed 2023-04-28 by shubnil