FreeRTOS/FreeRTOS-Plus-TCP #892: [BUG] Usage of uint64_t is against C90 standard

go2sh, 2023-05-31:
**Describe the bug**
long long and unsigned long long aka uint64_t are not part of the C90 standard. The tasking compiler will not compile the code when C90 is set as standard.

**Target**
- Toolchain and version: Tasking Tricore Compiler 6.4r1p6

**Host**
- Host OS: Windows
- Version: 10

**To Reproduce**
- Compile FreeRTOS-Plus-TCP with Tasking compiler

**Expected behavior**
Should compile

To use uint64_t standard conform the c standard has to increase to C99.

go2sh, 2023-06-01:
uint64_t is not part of the file you mentioned. The main problem with the tasking compiler is that `long long` and `unsigned long long` types, which are 64 bit in the target abi, are not available in C90 mode of the compiler. The reason is that the `long long` types were introduced in C99. The solution is easy for me, change the compile standard to C99 in the cmake target and the compilation works fine including the compiler provided uint64_t type.

I just wanted to point out, that a strict C90 compiler might not be able to compile the code.

go2sh, 2023-06-01:
It's in [FreeRTOS_IP_Private.h#L541](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/dev/IPv6_integration/source/include/FreeRTOS_IP_Private.h#L541) and 
[FreeRTOS_IP_Private.h#L728](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/dev/IPv6_integration/source/include/FreeRTOS_IP_Private.h#L728).

htibosch, 2023-06-01:
Ah yes, I thought so. I think that it wouldn't hurt anyone to remove those declarations.

moninom1, 2023-06-12:
@go2sh 
#907 Added fix for the Bug. Thank you for reporting it.

closed 2023-06-13 by moninom1