FreeRTOS/FreeRTOS-Plus-TCP #930: Unused function and unused variable warnings when ipconfigUSE_TCP_WIN is disabled

timaydin, 2023-06-18:
Testing with version 3.1.0

When ipconfigUSE_TCP_WIN is disabled, the following warnings occur:

../src/FreeRTOS-Plus-TCP/source/FreeRTOS_TCP_Transmission.c:1317:20: warning: unused variable 'xSizeWithoutData' [-Wunused-variable]

../src/FreeRTOS-Plus-TCP/source/FreeRTOS_TCP_WIN.c:386:17: warning: 'vListInsertGeneric' defined but not used [-Wunused-function]

kar-rahul-aws, 2023-06-19:
Hi @timaydin 
Thank you for reporting the issue. We are looking into it and will update after an internal discussion.

kar-rahul-aws, 2023-06-21:
Hi @timaydin 
We have added this [patch](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/932) to remove the warnings , on disabling the macro **ipconfigUSE_TCP_WIN**
We request you to check running your project with the latest change, and see if that fixes the issue reported.
Thanks.

moninom1, 2023-06-23:
Closing the issue with the PR being merged : https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/932
Feel free to let us know if you face any issue.

closed 2023-06-23 by moninom1