FreeRTOS/FreeRTOS-Plus-TCP #958: [BUG] Compiler errors/warnings - always have to manually ignore when building the Kernel with Clang

phelter, 2023-07-08:
**Describe the bug**
When compiling with the CMake options:

```cmake
add_compile_options(
    $<$<COMPILE_LANG_AND_ID:C,GNU>:-fdiagnostics-color=always>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-fcolor-diagnostics>

    $<$<COMPILE_LANG_AND_ID:C,Clang,GNU>:-Wall>
    $<$<COMPILE_LANG_AND_ID:C,Clang,GNU>:-Wextra>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wpedantic> # Note not used in GNU to allow C99 designated initializers.
    $<$<COMPILE_LANG_AND_ID:C,Clang,GNU>:-Werror>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Weverything>

    # Allow clang and gnu pramgmas.
    $<$<COMPILE_LANG_AND_ID:C,GNU>:-Wno-unknown-pragmas>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-typedef-redefinition>

    # Documentation types
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-fcomment-block-commands=retval> # Doesn't recongize doxygen retval
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-fcomment-block-commands=copydetails> # Doesn't recongize doxygen copydetails
)
```

There are several isssues/warnings that arise when compling with Clang-14 when compiling V4.0.0-rc3

**Target**
- Development board: N/A - Linux Posix
- Instruction Set Architecture: arm64
- IDE and version: None - cmake build no IDE.
- Toolchain and version: clang-14.0.0-1ubuntu1

**Host**
- Host OS: Ubuntu
- Version: 22.04

**To Reproduce**
- Use project - any and configure with above `add_compile_options`
- Build on ubuntu and could observe the following :

```
freertos_plus_tcp-src/source/FreeRTOS_BitConfig.c:135:15: error: parameter 'pxConfig:' not found in the function declaration [-Werror,-Wdocumentation]
freertos_plus_tcp-src/source/FreeRTOS_BitConfig.c:136:15: error: parameter 'pucData:' not found in the function declaration [-Werror,-Wdocumentation]
freertos_plus_tcp-src/source/FreeRTOS_BitConfig.c:137:15: error: parameter 'uxSize:' not found in the function declaration [-Werror,-Wdocumentation]
freertos_plus_tcp-src/source/FreeRTOS_BitConfig.c:203:59: error: implicit conversion loses integer precision: 'int' to 'uint16_t' (aka 'unsigned short') [-Werror,-Wimplicit-int-conversion]
freertos_plus_tcp-src/source/FreeRTOS_DHCP.c:785:17: error: default label in switch which covers all enumeration values [-Werror,-Wcovered-switch-default]
freertos_plus_tcp-src/source/FreeRTOS_DHCP.c:1375:23: error: mixing declarations and code is a C99 extension [-Werror,-Wdeclaration-after-statement]
freertos_plus_tcp-src/source/FreeRTOS_DNS_Callback.c:154:62: error: implicit conversion loses integer precision: 'TickType_t' (aka 'unsigned long') to 'uint32_t' (aka 'unsigned int') [-Werror,-Wshorten-64-to-32]
freertos_plus_tcp-src/source/FreeRTOS_DNS_Callback.c:276:10: error: no previous prototype for function 'vDNSCallbackInitialise' [-Werror,-Wmissing-prototypes]
freertos_plus_tcp-src/source/FreeRTOS_DNS_Callback.c:276:32: error: this old-style function definition is not preceded by a prototype [-Werror,-Wstrict-prototypes]
freertos_plus_tcp-src/source/FreeRTOS_DNS_Parser.c:474:60: error: implicit conversion loses integer precision: 'uint16_t' (aka 'unsigned short') to 'uint8_t' (aka 'unsigned char') [-Werror,-Wimplicit-int-conversion]
freertos_plus_tcp-src/source/FreeRTOS_DNS_Parser.c:563:98: error: empty expression statement has no effect; remove unnecessary ';' to silence this warning [-Werror,-Wextra-semi-stmt]
freertos_plus_tcp-src/source/FreeRTOS_DNS_Parser.c:575:122: error: implicit conversion loses integer precision: 'unsigned long' to 'int16_t' (aka 'short') [-Werror,-Wimplicit-int-conversion]
freertos_plus_tcp-src/source/FreeRTOS_DNS_Parser.c:582:143: error: empty expression statement has no effect; remove unnecessary ';' to silence this warning [-Werror,-Wextra-semi-stmt]
freertos_plus_tcp-src/source/FreeRTOS_DNS_Parser.c:1220:80: error: empty expression statement has no effect; remove unnecessary ';' to silence this warning [-Werror,-Wextra-semi-stmt]
freertos_plus_tcp-src/source/FreeRTOS_DNS_Parser.c:1224:122: error: empty expression statement has no effect; remove unnecessary ';' to silence this warning [-Werror,-Wextra-semi-stmt]
freertos_plus_tcp-src/source/FreeRTOS_IP.c:236:1: error: function 'prvIPTask' could be declared with attribute 'noreturn' [-Werror,-Wmissing-noreturn]
freertos_plus_tcp-src/source/FreeRTOS_IP.c:478:9: error: default label in switch which covers all enumeration values [-Werror,-Wcovered-switch-default]
freertos_plus_tcp-src/source/FreeRTOS_IP.c:893:23: error: mixing declarations and code is a C99 extension [-Werror,-Wdeclaration-after-statement]
freertos_plus_tcp-src/source/FreeRTOS_IP.c:1227:48: error: cast from 'const void *' to 'unsigned char *' drops const qualifier [-Werror,-Wcast-qual]
freertos_plus_tcp-src/source/FreeRTOS_IP.c:1660:9: error: default label in switch which covers all enumeration values [-Werror,-Wcovered-switch-default]
freertos_plus_tcp-src/source/FreeRTOS_IP.c:1888:25: error: variable 'uxHeaderLength' may be uninitialized when used here [-Werror,-Wconditional-uninitialized]
freertos_plus_tcp-src/source/FreeRTOS_IP_Utils.c:1030:47: error: mixing declarations and code is a C99 extension [-Werror,-Wdeclaration-after-statement]
freertos_plus_tcp-src/source/FreeRTOS_IP_Utils.c:67:13: error: macro is not used [-Werror,-Wunused-macros]
freertos_plus_tcp-src/source/FreeRTOS_IP_Utils.c:75:13: error: macro is not used [-Werror,-Wunused-macros]
freertos_plus_tcp-src/source/FreeRTOS_IPv6.c:640:16: error: variable 'xExtHeaderCount' set but not used [-Werror,-Wunused-but-set-variable]
freertos_plus_tcp-src/source/FreeRTOS_IPv4.c:420:86: error: unused parameter 'pxNetworkBuffer' [-Werror,-Wunused-parameter]
freertos_plus_tcp-src/source/FreeRTOS_DNS.c:933:33: error: enumeration value 'xPreferenceNone' not explicitly handled in switch [-Werror,-Wswitch-enum]
freertos_plus_tcp-src/source/FreeRTOS_DNS.c:1011:42: error: mixing declarations and code is a C99 extension [-Werror,-Wdeclaration-after-statement]
freertos_plus_tcp-src/source/FreeRTOS_DNS.c:1031:49: error: mixing declarations and code is a C99 extension [-Werror,-Wdeclaration-after-statement]
[... 101 lines cut ...]
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-gnu-statement-expression>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-implicit-int-conversion>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-missing-noreturn>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-missing-prototypes>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-shorten-64-to-32>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-sign-conversion>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-strict-prototypes>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-switch-enum>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-unused-but-set-variable>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-unused-macros>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-unused-parameter>
)

target_compile_options(freertos_plus_tcp_network_if
  PRIVATE
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-cast-qual>
)
```

**Screenshots**
N/A
**Wireshark logs**
N/A
**Additional context**
N/A

phelter, 2023-07-11:
@moninom1 I am also seeing the following issues when compiling with 
gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0 
And RelWithDebInfo settings - wtih above add_compile_options.

```
./freertos_plus_tcp-src/source/include/FreeRTOS_IP_Private.h: In function 'vSocketClose':
./freertos_plus_tcp-src/source/FreeRTOS_Sockets.c:2304:29: error: array subscript 'ListItem_t {aka const struct xLIST_ITEM}[0]' is partly outside array bounds of 'List_t[1]' {aka 'struct xLIST[1]'} [-Werror=array-bounds]
 2304 |             for( pxIterator = listGET_NEXT( pxEnd );
      |                             ^
In file included from ./freertos_plus_tcp-src/source/include/FreeRTOS_IP_Utils.h:57,
                 from ./freertos_plus_tcp-src/source/include/FreeRTOS_IP.h:495,
                 from ./freertos_plus_tcp-src/source/include/FreeRTOS_UDP_IP.h:41,
                 from ./freertos_plus_tcp-src/source/FreeRTOS_Sockets.c:46:
./freertos_plus_tcp-src/source/include/FreeRTOS_IP_Private.h:363:19: note: while referencing 'xBoundTCPSocketsList'
  363 |     extern List_t xBoundTCPSocketsList;
      |                   ^~~~~~~~~~~~~~~~~~~~
./freertos_plus_tcp-src/source/FreeRTOS_Sockets.c:2280:24: error: array subscript 'ListItem_t {aka const struct xLIST_ITEM}[0]' is partly outside array bounds of 'List_t[1]' {aka 'struct xLIST[1]'} [-Werror=array-bounds]
 2280 |             pxIterator = listGET_NEXT( pxEnd );
      |                        ^
In file included from ./freertos_plus_tcp-src/source/include/FreeRTOS_IP_Utils.h:57,
                 from ./freertos_plus_tcp-src/source/include/FreeRTOS_IP.h:495,
                 from ./freertos_plus_tcp-src/source/include/FreeRTOS_UDP_IP.h:41,
                 from ./freertos_plus_tcp-src/source/FreeRTOS_Sockets.c:46:
./freertos_plus_tcp-src/source/include/FreeRTOS_IP_Private.h:363:19: note: while referencing 'xBoundTCPSocketsList'
  363 |     extern List_t xBoundTCPSocketsList;
      |                   ^~~~~~~~~~~~~~~~~~~~
./freertos_plus_tcp-src/source/include/FreeRTOS_IP_Private.h: In function 'pxUDPSocketLookup':
./freertos_plus_tcp-src/source/FreeRTOS_Sockets.c:3039:25: error: array subscript 'ListItem_t {aka const struct xLIST_ITEM}[0]' is partly outside array bounds of 'List_t[1]' {aka 'struct xLIST[1]'} [-Werror=array-bounds]
 3039 |         for( pxIterator = listGET_NEXT( pxEnd );
      |                         ^
In file included from ./freertos_plus_tcp-src/source/include/FreeRTOS_IP_Utils.h:57,
                 from ./freertos_plus_tcp-src/source/include/FreeRTOS_IP.h:495,
                 from ./freertos_plus_tcp-src/source/include/FreeRTOS_UDP_IP.h:41,
                 from ./freertos_plus_tcp-src/source/FreeRTOS_Sockets.c:46:
./freertos_plus_tcp-src/source/include/FreeRTOS_IP_Private.h:325:15: note: while referencing 'xBoundUDPSocketsList'
  325 | extern List_t xBoundUDPSocketsList;
      |               ^~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

Suggest running with gcc installed in Ubuntu 22.04 - (v 11.2 or above) and run with release builds.

moninom1, 2023-07-27:
Hi @phelter I have fixed the clang issues, but i was not able to see any issue with gcc on Ubuntu. I am working on FreeRTOS-Plus-TCP main branch as we have merged all the dev/IPv6_Integration chages back to main.

phelter, 2023-07-27:
@moninom1 - are you running with the latest GNU GCC compiler or at a minimum v11.2 or higher?
It is possible that those files have changed since merging the IPv6 integration branch into master, and those issues were fixed by the other branch.

moninom1, 2023-07-29:
Hi @phleter, Yes I am using this version - 
gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0

Can you please help in checking on the latest FreeRTOS-Plus- TCP main branch, is the issues are not there, we can conclude that they are resolved.

moninom1, 2023-08-01:
Hi @phelter 
As the Clang changs are merged [PR984](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/984) i will close the issue for now. Feel free to create new issue if you find any issue with gcc. One again thank you for reporting the issue to us.

closed 2023-08-01 by moninom1