FreeRTOS/FreeRTOS-Plus-TCP #959: [BUG] Improper use of headers and includes.

phelter, 2023-07-08:
**Describe the bug**
The way headers are included they have to be in a particular order and not all headers that define types and parameters are added to the headers of others.

Also the:

```c
    #ifdef __cplusplus
        extern "C" {
    #endif
```
should be *AFTER* any other includes in the header and not before.  Otherwise if a header is missing the extern-C you may have 2 definitions of the same function (one for extern-C style) and one for native Cpp mangled name.

**Target**
- Development board: None - Linux
- Instruction Set Architecture: x86/amd64
- IDE and version: N/A - cmake builds
- Toolchain and version: clang-14

**Host**
- Host OS: Ubuntu
- Version: 22.04

**To Reproduce**
- Use project where you have the following include order:
```c
/* FreeRTOS+TCP includes. */
#include "FreeRTOS_DHCP.h"
#include "FreeRTOS_IP.h"
#include "FreeRTOS_Sockets.h"
```

- Attempt to compile and observe errors where used:
```
freertos_plus_tcp-src/source/include/FreeRTOS_DHCPv6.h:145:5: error: unknown type name 'eDHCPState_t'
freertos_plus_tcp-src/source/include/FreeRTOS_Routing.h:51:65: error: unknown type name 'NetworkBufferDescriptor_t'
freertos_plus_tcp-src/source/include/FreeRTOS_Routing.h:159:9: error: unknown type name 'MACAddress_t'; did you mean 'IP_Address_t'?
freertos_plus_tcp-src/source/include/FreeRTOS_Routing.h:179:13: error: unknown type name 'IPTimer_t'
freertos_plus_tcp-src/source/include/FreeRTOS_Routing.h:182:13: error: unknown type name 'DHCPData_t'
freertos_plus_tcp-src/source/include/FreeRTOS_Routing.h:246:59: error: unknown type name 'MACAddress_t'; did you mean 'IP_Address_t'?
freertos_plus_tcp-src/source/include/FreeRTOS_Routing.h:288:60: error: use of undeclared identifier 'ipIP_ADDRESS_LENGTH_BYTES'
freertos_plus_tcp-src/source/include/FreeRTOS_Routing.h:289:58: error: use of undeclared identifier 'ipIP_ADDRESS_LENGTH_BYTES'
freertos_plus_tcp-src/source/include/FreeRTOS_Routing.h:290:65: error: use of undeclared identifier 'ipIP_ADDRESS_LENGTH_BYTES'
freertos_plus_tcp-src/source/include/FreeRTOS_Routing.h:291:67: error: use of undeclared identifier 'ipIP_ADDRESS_LENGTH_BYTES'
freertos_plus_tcp-src/source/include/FreeRTOS_Routing.h:292:61: error: use of undeclared identifier 'ipMAC_ADDRESS_LENGTH_BYTES'
freertos_plus_tcp-src/source/include/FreeRTOS_Routing.h:303:70: error: use of undeclared identifier 'ipMAC_ADDRESS_LENGTH_BYTES'
```

Please add what each header uses or provide an explicit documentation on the requirements of header ordering for this project to work.

**Expected behavior**
Any header can be added in any order and it will work.  Using header guards for ensuring something doesn't get added twice.

**Screenshots**
N/A

**Wireshark logs**
N/A

**Additional context**
N/A

moninom1, 2023-07-10:
HI @phelter,
Thank you for bringing this to our attention! We will be looking further into this issue as soon as we can, thank you for your patience!”

moninom1, 2023-07-25:
Hi @phelter,
We looked into the issue and here are our observations : 
1.  Yeah extern “C” block can be moved after the #includes and we will create a PR for the same.
2. For order of header includes,, we see some files have the issue and we working on finding an efficient way to fix it.

moninom1, 2023-07-31:
Hi @phelter, Both the issue's PR are merged, hence closing the PR. Thank you for reporting the issue.

closed 2023-07-31 by moninom1