FreeRTOS/FreeRTOS-Plus-TCP #962: [Feature Request] Multicast Support

Hunter-Phipps, 2023-07-11:
Hi, could you please add Multicast support.

shubnil, 2023-07-13:
Hi @Hunter-Phipps ,
Thanks for asking.
Just for clarification , is the ask for Multicast support for IPv4 or IPv6.

htibosch, 2023-08-22:
Hello @Hunter-Phipps, I see that your issue is getting little response.

Currently there is a very limited use of multicast: only in the mDNS and LLMNR protocols.

Another popular application is IGMP. @evpopov has been active with it, as you [must have seen](https://forums.freertos.org/t/tcp-ipv4-multicasts-and-igmp/11934).

For what purpose do you need multicast?

Using **multicast** is a bit like UDP in **broadcast** mode. A sockets is bound to a port number, and when a packet is received for that port number, it will be received on that socket.

One thing to do is set filtering of the MAC- and IP-addresses:

- At the network interface should be told that certain multicast MAC-addresses should be let through. The EMAC often has a has hash register for this.
- The IP-addresses should also be allowed in functions like `prvAllowIPPacketIPv4()` / `prvAllowIPPacketIPv6()`.

evpopov, 2023-08-22:
For everyone interested, you can follow my branch here: [MCast branch](https://github.com/evpopov/FreeRTOS-Plus-TCP/tree/MCast) just please don't take anything for granted. I rebase and force-push this branch every once in a while to keep it based off of the latest "main".
As @htibosch mentioned earlier, supporting multicasts requires that the hardware be told which multicast addresses it needs to  allow in. In the hardware that I'm working with ( SAME70 ) you can have up to 4 individual addresses + a hash table.
@Hunter-Phipps, if you decide to use the branch, you will have to provide your own add/remove functions and assign them to the interface... something like this:
```
NetworkInterface_t * pxSAM_FillInterfaceDescriptor( BaseType_t xEMACIndex,
                                                    NetworkInterface_t * pxInterface )
{
    .
    .
    .
	#if ( ipconfigSUPPORT_IP_MULTICAST != 0 )
		pxInterface->pfAddMulticastMAC = prvAddMulticastMACAddress;
		pxInterface->pfRemoveMulticastMAC = prvRemoveMulticastMACAddress;
	#endif
    .
    .
    .
}
```
Or you can always assign functions that don't do anything as a start and just configure your hardware to receive all multicasts.

@htibosch, you may be interested in how I did my DriverSAM/NetworkInterface.s and in particular the adding and removal of addresses. It attempts to keep track of how many sockets are using a particular hash and only remove the hash when no more sockets need it.
@htibosch, do you think I should just do a "work in progress PR" for this or do you think keeping it separate like it is now is cleaner.

hangeV, 2023-09-03:
> Hello @Hunter-Phipps, I see that your issue is getting little response.
>
> Currently there is a very limited use of multicast: only in the mDNS and LLMNR protocols.
>
> Another popular application is [...]

Multicast is necessary, such as the Rockwell EtherNet/IP protocol, a popular industrial Ethernet protocol

archigup, 2023-09-21:
Hi, this feature request is under consideration