FreeRTOS/FreeRTOS-Plus-TCP #967: Use of apostrophe in a #warning message causes build error

sidmodi-mw, 2023-07-13:
**Describe the bug**
A `#warning` message used in [Zynq/NetworkInterfaces.c](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/0d64b4395d1118985a4ade5475e3c865cb7afe1a/source/portable/NetworkInterface/Zynq/NetworkInterface.c#L330C1-L331C1) uses the word `doesn't` and causes the build to fail with the error
```
error: missing terminating ' character
```

**Target**
- Development board: None - Linux
- Instruction Set Architecture: x86/amd64
- IDE and version: N/A - cmake builds
- Toolchain and version: arm-gnu-toolchain-12.2.mpacbti-rel1-x86_64

**Host**
- Host OS: Ubuntu
- Version: 22.04

**To Reproduce**
- Compile a project that uses the Zynq port and builds Zynq/NetworkInterfaces.c

**Expected behavior**
The build passes successfully

**Request**
- Ideally, this `#warning` would be replaced with a `#pragma message` since the warning is always going to be present and prevents the user from treating warnings as errors if they so desire.
```
- #warning As 'MCASTHASHEN' doesn't seem to work, use the promiscuous mode so that IPv6 multicast packets are received.
+ #pragma message ("As 'MCASTHASHEN' doesn't seem to work, use the promiscuous mode so that IPv6 multicast packets are received.")
```

ActoryOu, 2023-07-14:
Hi @sidmodi-mw,
Thanks for bringing this to our attention!

We're now working on the fix.
I'll update later when the PR is created.

Thanks.

sidmodi-mw, 2023-07-14:
@ActoryOu Thanks for the update! Quick note - I noticed that there are other `#warning` messages in there as well that connot be suppressed unless a certain configuration is used. Would it be possible to change them to `#pragma message` as well? I'm talking specifically about ones to do with `ipconfigDRIVER_INCLUDED_RX_IP_CHECKSUM` and `ipconfigDRIVER_INCLUDED_TX_IP_CHECKSUM` being disabled causes `warnings` but there might be more.

ActoryOu, 2023-07-17:
@sidmodi-mw, That's what we're discussing.
Could you describe more about why you're disabling `ipconfigDRIVER_INCLUDED_RX_IP_CHECKSUM` or `ipconfigDRIVER_INCLUDED_TX_IP_CHECKSUM`? I thought that it's not necessary to change these configurations in this platform.

sidmodi-mw, 2023-07-17:
Hi @ActoryOu, we had those flags disabled in our last release build (not sure why) but we do plan to enable it. However, I want to first run the new build without changing any configurations.

ActoryOu, 2023-07-18:
#971 created to provide solution to suppress warning messages.

ActoryOu, 2023-07-20:
I'm closing this issue cause #971 has merged.

closed 2023-07-20 by ActoryOu