FreeRTOS/FreeRTOS-Plus-TCP #1008: TCP zero-copy TX doesn't work.

evpopov, 2023-08-18:
From the scarce information about TCP zero copy transmit I was able to figure out that I need to get a pointer to the stream buffer, copy my data and call FreeRTOS_send() with a NULL pointer.
The issue is the code snippet below from FreeRTOS_send() :
```
        BaseType_t xByteCount = -pdFREERTOS_ERRNO_EINVAL;
        FreeRTOS_Socket_t * pxSocket = ( FreeRTOS_Socket_t * ) xSocket;

        if( pvBuffer != NULL )
        {
            /* Check if this is a valid TCP socket, affirm that it is not closed or closing,
             * affirm that there was not malloc-problem, test if uxDataLength is non-zero,
             * and if the connection is not in a confirmed FIN state. */
            xByteCount = ( BaseType_t ) prvTCPSendCheck( pxSocket, uxDataLength );
        }
        if( xByteCount > 0 )
        {
            /* prvTCPSendLoop() will try to send as many bytes as possible,
             * returning number of bytes that have been queued for transmission.. */
            xByteCount = prvTCPSendLoop( pxSocket, pvBuffer, uxDataLength, xFlags );

            if( xByteCount == 0 )
            {
                      .
                      .
                      . 
            }
        }

        return xByteCount;
```
This will never work with pvBuffer = NULL and I was always getting a return of -pdFREERTOS_ERRNO_EINVAL.
That leads to a question: Is the use of NULL pointer deprecated now? If it is still allowed, then the check above is not correct.

After commenting out the ` if( pvBuffer != NULL )` check I was able to send data in the "zero copy" way.

**Target**
- All

**Host**
- NA

**To Reproduce**
- call FreeRTOS_send() with a NULL buffer pointer

**Expected behavior**
FreeRTOS_send() should return the number of bytes sent.

htibosch, 2023-08-19:
Hi @evpopov, you are right, this doesn't work (anymore).
I will have a look at it and report back here.
Thanks for raising the issue.
Hein

htibosch, 2023-08-19:
The following seems to solve the problem:

~~~diff
-    BaseType_t xByteCount = -pdFREERTOS_ERRNO_EINVAL;
+    BaseType_t xByteCount;
     FreeRTOS_Socket_t * pxSocket = ( FreeRTOS_Socket_t * ) xSocket;
 
-    if( pvBuffer != NULL )
-    {
-        /* Check if this is a valid TCP socket, affirm that it is not closed or closing,
-         * affirm that there was not malloc-problem, test if uxDataLength is non-zero,
-         * and if the connection is not in a confirmed FIN state. */
-        xByteCount = ( BaseType_t ) prvTCPSendCheck( pxSocket, uxDataLength );
-    }
+    xByteCount = ( BaseType_t ) prvTCPSendCheck( pxSocket, uxDataLength );
 
     if( xByteCount > 0 )
~~~
I will test it more and create a PR for this.

evpopov, 2023-08-20:
Thanks @htibosch !
I'm glad the solutions is that easy, I just wasn't sure about side effects as I've never done zero-copy on a TCP socket before.
I'll close this. Thanks again.

closed 2023-08-20 by evpopov