FreeRTOS/FreeRTOS-Plus-TCP #1042: [BUG] Incorrect responses to name queries.

evpopov, 2023-11-10:
**Describe the bug**
In certain configuration scenarios, +TCP will respond to mDNS/LLMNR queries with a source address of 0.0.0.0 or will send the responses out of the wrong network interface.

**Target**
- Any

**To Reproduce**
[image]

**Wireshark logs**
[EndPointSearchIssue.zip](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/13323147/EndPointSearchIssue.zip)

**Detailed description**
In both examples, one could argue that the devices are misconfigured, but please bear with me.
In both examples, I'm executing "ping DUT" on a Windows 10 Pro PC. The device actually pings over IPv6, but there is weird stuff happening which got me to dig deeper. 
It should go without saying that in both examples, the DUT calls xApplicationDNSQueryHook_Multi() which returns pdTRUE and the stack tries to respond that it has the "DUT" name.

Single Interface Example:
In the single_interface.pcapng file, you can see that my PC tries to locate the IPv4 and IPv6 addresses of my DUT and the PC uses both LLMNR and mDNS. On IPv6, everything works fine, but the IPv4 responses have a source address of 0.0.0.0
After some debugging, the problem is  this:
https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/9abe2d11c6a8ee7025ba581eeed6d227ca3d93d7/source/FreeRTOS_DNS_Parser.c#L914C14-L914C14
```
NetworkEndPoint_t * pxEndPoint = prvFindEndPointOnNetMask( pxNetworkBuffer );
```
What is happening there is the stack has received a name query and wants to respond to it so the code searches for the appropriate end-point to use.
In my example, there is no end-point that matches the source IP of 169.254.254.253 and the result is NULL. Following the code further,....
```
pxIPHeader->ulSourceIPAddress = ( pxEndPoint != NULL ) ? pxEndPoint->ipv4_settings.ulIPAddress : 0U;
```

Dual Interface Example:
The offending code now finds and endpoint that is in the 169.254.0.0 network and sends the mDNS/LLMNR replies to it.... but wait, that is on a different network interface.
Looking at the Wireshark captures, dual_interface_pc_side.pcapng is missing the IPv4 responses, but they are present in the dual_interface_far_side.pcapng capture.

I believe everything described above is incorrect behavior. I believe the offending line above should be changed from
```
NetworkEndPoint_t * pxEndPoint = prvFindEndPointOnNetMask( pxNetworkBuffer );
```
to
```
NetworkEndPoint_t * pxEndPoint = pxNetworkBuffer->pxEndPoint;
```
Here is my reasoning:
When receiving frames, the stack does a pretty exhaustive search trying to locate the best end-point for every incoming packet. So this work was done during the reception of all mDNS/LLMNR queries. Why would we search again only to send the response? we know where the question came from.
And as you can see, searching for an end-point can actually create the weird situation where queries on one interface generate traffic on another.  

I have looked long and hard into my proposed solution and I cannot find any execution path that leads to pxNetworkBuffer->pxEndPoint being NULL. It always represents a good end-point and it is always the end-point that received the query. I have been running this fix on actual hardware for a few days now and I am quite positive there are no theoretical or practical ill effects but I did not want to rush creating a pull request for this.
With the proposed fix applied, the responses to the queries come out as expected. The reason these are valuable is because the DUT doesn't actually know what IP addresses the PC has. By sending valid responses to the proper multicast IP addresses, the PC finds out the IPv4 of the DUT and if the PC has a compatible IPv4, it may contact the DUT under IPv4

Please let me know what you think or if you think I'm missing something important.

Final note: I don't want to muddy the water, but if you look closely in the dual_interface_far_side.pcapng file, you may notice that the source MAC address of the DUT is the same as the network interface that faces the PC. This is incorrect and I believe is another bug that is the result of:
https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/9abe2d11c6a8ee7025ba581eeed6d227ca3d93d7/source/FreeRTOS_IP.c#L2120C30-L2120C30
but let's not get into that yet.

cookpate, 2023-11-11:
Thanks for bringing this to our attention. We will look further into this issue as soon as we can.

htibosch, 2023-11-13:
Hi @evpopov, thanks for reporting these issues.

I can still not reproduce the problem. Can you produce a description of the interface/endpoint configuration.

Here is mine, a test configuration for mDNS:

~~~

Interface eth0

Laptop:
IP-address : 192.168.2.11

IP-address : 192.168.2.107
End-point  : up = yes method static
Net mask   : 255.255.255.0
GW         : 192.168.2.1
DNS-0      : 118.98.44.10
DNS-1      : 118.98.44.100
DNS-2      : 0.0.0.0
Broadcast  : 192.168.2.255
MAC address: 00-11-11-11-11-42
 
IP-address : 172.16.0.107
End-point  : up = yes method static
Net mask   : 255.255.224.0
GW         : 172.16.0.1
DNS-0      : 0.0.0.0
DNS-1      : 0.0.0.0
DNS-2      : 0.0.0.0
Broadcast  : 172.16.31.255
MAC address: 00-11-11-11-11-07

IP-address : 2600:70ff:c066::2001
End-point  : up = yes method static
Prefix     : 2600:70ff:c066::/64
GW         : fe80::ba27:ebff:fe5a:d751
DNS-0      : fe80::1
DNS-1      : 2001:4860:4860::8888
DNS-2      : ::
MAC address: 00-11-11-11-11-42
 
IP-address : fe80::7004
End-point  : up = yes method static
Prefix     : fe80::/10
GW         : ::
DNS-0      : ::
DNS-1      : ::
DNS-2      : ::
MAC address: 00-11-11-11-11-42
~~~
Can you share your complete configuration?
And which platform are you using, SAMxx?
Thanks.

evpopov, 2023-11-13:
@htibosch 
In your list of end points, I'm not sure which one is sends the mDNS queries and which endpoints are on the device-under-test.

The simplest way to reproduce it should be the single interface example depicted above. The important thing is to have your PC be in a different IPv4 subnet than the device under test. The picture above shows exactly what my setup is on IPv4 and ignores the IPv6 end-points.
My laptop is manually set to 169.254.254.253/16 but I could have also used 10.x.x.x/8 or 192.168.x.x/16
Here is what `ipconfig` gives me on my Windows 10 ethernet settings:
```
Ethernet adapter Ethernet:

   Connection-specific DNS Suffix  . :
   Link-local IPv6 Address . . . . . : fe80::42cc:f99e:3526:52ce%13
   IPv4 Address. . . . . . . . . . . : 169.254.254.253
   Subnet Mask . . . . . . . . . . . : 255.255.0.0
   Default Gateway . . . . . . . . . :
```

My SAME70 based device only has one IPv4 end-point set for a static 172.30.30.26/24
```
Interface  : GMAC0
    IP-Address : 172.30.30.26
    End-point  : up = yes method static
    Net mask   : 255.255.255.0
    GW         : 172.30.30.1
    DNS        : 1.1.1.1
    MAC address: 70-B3-D5-09-C0-06

    IP-Address : fe80::72b3:d5ff:fe09:c006
    End-point  : up = yes method static
    Prefix     : fe80::/10
    GW         : ::
    DNS 0      : ::
    DNS 1      : ::
    MAC address: 70-B3-D5-09-C0-06
```

Again, the important part is that **the laptop and the SAME70 be on different subnets**.
The mDNS packets from the laptop are sent to 224.0.0.251 and the SAME70 receives them on it's only IPv4 endpoint: the one that is set to 172.30.30.26. Everything goes as expected, xApplicationDNSQueryHook_Multi() gets called and returns pdTRUE. 
A few lines later prepareReplyDNSMessage() gets called which internally calls prvFindEndPointOnNetMask()
prvFindEndPointOnNetMask searches for and end-point that corresponds to the source address of the mDNS query. This source address is 169.254.254.253 and there is no end-point that can satisfy the search, so prvFindEndPointOnNetMask() returns NULL.
You should see the IPv4 mDNS responses have source IP of 0.0.0.0

I'm attaching my config file as well.
[FreeRTOSIPConfig.zip](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/13336075/FreeRTOSIPConfig.zip)

Let me know if that helps.

htibosch, 2023-11-17:
You wrote:
> EDIT: Please remove your 192.168.2.107 end-point and try again.

Indeed, that helps, now I get an answer with the 0.0.0.0 return address.

One more question: what's the use of mixing networks on the same LAN?

Laptop: 169.254.254.253/16
DUT: 172.30.30.26/24

because even if the DUT sends an answer with 172.30.30.26 as a return address, the laptop will use a gateway to try to reach the DUT.

evpopov, 2023-11-17:
In many cases, using multiple networks on a LAN is an indication of a problem, but keep in mind that there is nothing wrong or illegal with having multiple IP addresses assigned to an interface. In fact, I have not heard of any RFC or standard that prohibits it, so yes, it is a niche case, but it's perfectly possible. I also have to note that sending responses with a source address of 0.0.0.0 is also not particularly "illegal" either.

Consider the following scenario:
Dev-PC: 192.168.0.1/24, 172.16.0.1/16
DUT: 172.16.0.2
We don't really know which IP address the PC will use when sending mDNS queries. Imagine that it just so happens to use 192.168.0.1 The DUT should still respond with 172.16.0.2 and then the PC will actually be able to connect without using a gateway. Weird scenario, I know... but no RFC or standard forbids it and it's quite possible. Currently, my Windows PC seems to ignore the mDNS responses when they are sent from 0.0.0.0. It shouldn't be ignoring them, but it does and I don't know why.

A slightly more concerning situation is if this behavior is used for malicious purposes.
Imagine a DUT with multiple interfaces. A bad actor on the Ethernet side of the DUT can easily generate queries with a source address that is different than what is used on the Ethernet side of the DUT. The responses to these will either be sent on the Ethernet side with source IP of 0.0.0.0 or they will be sent on the other interface. This depends on the source address used in the query. This poses two issues:
1 - The bad actor can thus figure out the network of the other interface of the DUT because he/she will not see responses on the Ethernet side when they use a specific source IP. The lack of responses means the DUT has another network interface that belongs to the source IP that the bad actor used.
2 - The second interface may be much lower speed than standard Ethernet and a large amount of queries on the Ethernet side can generate a DoS condition on the other interface.

I admit, the issues that I'm describing are not "the end of the world" and are a bit academic, but my main point is that they would not exist if we keep it simple: An end-point has already been selected. Let's use it, instead of wasting CPU to only create more issues. The search for an end-point [here](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/9abe2d11c6a8ee7025ba581eeed6d227ca3d93d7/source/FreeRTOS_DNS_Parser.c#L914C14-L914C14) doesn't do any good. It doesn't solve a problem. It doesn't add features. It just adds code, wastes CPU time, and raises questions. 
That's all I'm trying to say.

Thanks for spending some time to look into this Hein.

htibosch, 2023-11-28:
Hi @evpopov, I'm sorry that my response took so long. Now I that I can replicate it, I looked at your diagnosis, changed it, and ran more tests, with success.

I implemented your proposed change like this:
~~~diff
 void prepareReplyDNSMessage( NetworkBufferDescriptor_t * pxNetworkBuffer,
                              BaseType_t lNetLength )
 {
     UDPPacket_t * pxUDPPacket;
     IPHeader_t * pxIPHeader;
     UDPHeader_t * pxUDPHeader;
     size_t uxDataLength;
-    NetworkEndPoint_t * pxEndPoint = prvFindEndPointOnNetMask( pxNetworkBuffer );
+    NetworkEndPoint_t * pxEndPoint = pxNetworkBuffer->pxEndPoint;
     const size_t uxIPHeaderLength = uxIPHeaderSizePacket( pxNetworkBuffer );
 
+    configASSERT( pxEndPoint != NULL );
     pxUDPPacket = ( ( UDPPacket_t * )
~~~

evpopov, 2023-11-28:
Thanks @htibosch.
I like the idea of adding an assert call there. I'll leave this issue open for now in case someone else wants to chime in.
I still can't find a case where the proposed change can result in something bad, so if nothing  changes, I plan on closing the issue in about a week and opening a PR with those exact 2 changes + a reference to this discussion.

closed 2023-12-20 by evpopov