FreeRTOS/FreeRTOS-Plus-TCP #1070: [Feature Request] <add BSD sockets porting>

bigtreeboy, 2024-01-04:
I have had a BSD sockets based application.
Now I want to change the TCP/IP stack to freeRTOS-Plus-TCP.
but, unfortunately, freeRTOS-Plus-TCP doesn't supply the BSD sockets interface directly.
If freeRTOS-Plus-TCP has the BSD sockets interface, I don't need change anything in the application.

tony-josi-aws, 2024-01-04:
Hi @bigtreeboy,

FreeRTOS+TCP supports  BSD style sockets and APIs. Refer: [Primary Sockets Functions](https://www.freertos.org/FreeRTOS-Plus/FreeRTOS_Plus_TCP/FreeRTOS_TCP_API_Functions.html)

Suggest you to post these queries in the [FreeRTOS Forum](https://forums.freertos.org/).

bigtreeboy, 2024-01-05:
thanks for your answer.
But, was there these header file in current freeRTOS-Plus-TCP repo?
according to [Berkeley sockets](https://en.wikipedia.org/wiki/Berkeley_sockets), the standard headers are sockets.h netdb.h and so on. 
Because I have had a unix based application, I want reuse this code without changing anything.

htibosch, 2024-01-05:
PS. please close this subject and continue on the [FreeRTOS forum](https://forums.freertos.org).

What you can do is create a new set of sockets.h, netdb.h etc, and from there you can include the FreeRTOS includes.

Also, you can introduce these kind of macros:
~~~
#define socket    FreeRTOS_socket
#define bind      FreeRTOS_bind
#define listen    FreeRTOS_listen
#define connect   FreeRTOS_connect
#define accept    FreeRTOS_accept
#define recvfrom  FreeRTOS_recvfrom
#define sendto    FreeRTOS_sendto
#define recv      FreeRTOS_recv
#define send      FreeRTOS_send
~~~
You will have to study the Linux application and check if the FreeRTOS functions do exactly the same.
And at some point, you will need an intermediate function when the use of parameters is slightly different, such as `select()`.

bigtreeboy, 2024-01-05:
If there is a official version for BSD sockets interface, I think it's better.
of cause, I can implement it by myself.
I will close this feature request.

closed 2024-01-05 by bigtreeboy