FreeRTOS/FreeRTOS-Plus-TCP #1077: [BUG] The endpoint with the NULL address is operated in prvTCPReturnPacket_IPV4 function

XIAO-Q7, 2024-01-19:
**Describe the bug**

When a endpoint enables DHCP and sets a AUTO IP. Receiving a tcp socket packet from a non-AUTOIP network segment causes the endpoint with a null address to be operated on in the prvTCPReturnPacket_IPV4 function.

**Target**

Any

**To Reproduce**
[image]

- Configure: An endpoint enable DHCP and set a AUTO IP

In my case, I'm running Plus TCP on an STM32 and built a TCP echo server, employing a computer as the echo client. When I send an echo frame to the server, the microcontroller runs into HardFault_Handler.

**Expected behavior**

Server correctly returns echo frames.

I have set a special IP address for the computer in this case, not the AUTOIP segment. If PLUS TCP can't automatically find an entry point to send frames. It can not send the frame. I don't want the chip to enter HardFault_Handler.

**Additional context**

Through my debugging, I found that in the prvTCPReturnPacket_IPV4 function, the line of code here gives an endpoint to a null address.

```C
eResult = eARPGetCacheEntry( &ulDestinationIPAddress, &xMACAddress, &( pxNetworkBuffer->pxEndPoint ) );
```

and later operates this endpoint with a null address to send frames, eventually causing the chip to run into HardFault_Handler.

```C
NetworkInterface_t * pxInterface = pxNetworkBuffer->pxEndPoint->pxNetworkInterface;
( void ) pxInterface->pfOutput( pxInterface, pxNetworkBuffer, xDoRelease );
```

htibosch, 2024-01-19:
Hello @XIAO-Q7 , thanks a lot for reporting this.
I'm trying to reproduce it, and a fix is on its way.
Thanks

tony-josi-aws, 2024-01-29:
@XIAO-Q7, we have merged #1078 that should be fixing the issue reported, if possible, please verify if the fix works for you by pulling the latest changes from main.

XIAO-Q7, 2024-01-30:
Yes, it worked. Thank you for your help.

closed 2024-01-30 by XIAO-Q7

tony-josi-aws, 2024-01-30:
Thanks for verifying.