FreeRTOS/FreeRTOS-Plus-TCP #1093: [BUG]prvProcessICMPMessage_IPv6() sends replies to the wrong Endpoint

evpopov, 2024-02-05:
**Describe the bug**
As as already discussed in #1064, I found the same type of issue when handling ICMPv6 ND packets.

**Target**
- ALL

**To Reproduce**
In my test setup I have two network interfaces. One has an IPv6 Endpoint and is connected to my PC. The second interface is actually a fake interface that pretends to be UP and is not connected to anything. At this stage, My PC has no problem pinging the first interface by it's link-local IPv6 address.
The issue arises when I configure an IPv6 link-local Endpoint **on the second interface**. As soon as I do that, I'm no longer able to ping the first interface from my PC.
What happens is, the PC sends a Neighbor Solicitation packet to the first interface, FreeRTOS+TCP receives it and sends the response out the second interface.
The problematic piece of code is: [HERE](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/6310e61e9430cb2d50b1c128c9c00b06ba96ecd4/source/FreeRTOS_ND.c#L1060)
As was the case in #1064, we are searching for an endpoint when we already have a known good one.
Changing
```
NetworkEndPoint_t * pxEndPointFound = FreeRTOS_FindEndPointOnIP_IPv6( &( pxICMPHeader_IPv6->xIPv6Address ) );
```
to
```
NetworkEndPoint_t * pxEndPointFound = pxNetworkBuffer->pxEndPoint;
```
fixes the issue.

rant-mode-on....
I keep finding these small little issues and it is starting to feel like FreeRTOS+TCP might need some kind of a review of the general approach to how Endpoints are handled. When do we search for endpoints vs when do we trust the endpoint that is already selected. When we search for endpoint, do we just search based on address/netmask/prefix or do we need to also consider the network interface? The deeper I dig, the more these questions pop up. I don't want to water down this specific issue, but I have another similar endpoint-related issue in the works and it again involves one of these "FindEndpointXXXXX" functions. Something just doesn't feel right. The deeper I dig into +TCP, the more weird issues keep popping up and to be honest, I don't feel completely comfortable with the solutions that I'm proposing. This code has been written by people that have a lot more experience than me and I don't believe I'm competent enough to correct their decisions, but at the same time, I have a job to do and a device that needs to work properly, so I have to fix these issue. And if I don't push my changes upstream, I end up with too many changes that I need to rebase every time "main" pushes forward.

Please excuse the rant above but I needed to share my frustration.
FreeRTOS+TCP is amazing and I love working with it and on it. The IPv6 functionality is a literal life saver for me and I really really really want it to work properly. 
Thanks for your time.

paulbartell, 2024-02-05:
Thanks for the bug report. Our team will take a look.

shubnil, 2024-02-06:
Thanks @evpopov for bringing this out. We are looking into the same and will get back on this soon.

Thanks,
Shub

shubnil, 2024-02-07:
Hi @evpopov , as you suggested we can use the endpoint of the NetworkBuffer to to map to the correct Link Local endpoint. Will you be creating a change or we can also create one. Please suggest.

evpopov, 2024-02-07:
@shubnil This on seems pretty straight-forward. I'll create a PR. I will appreciate any help from the team if it fails the CI tests

closed 2024-02-07 by evpopov