FreeRTOS/FreeRTOS-Plus-TCP #1101: Compile problem with RP2040 and PROPERTY C_STANDARD 90

wdalmut, 2024-02-10:
**Describe the bug**
when you try to compile the FreeRTOS Plus TCP (V4) with RP2040 the build starts failing on pico library  with multiple errors like:

```
expected declaration specifiers or '...' before string constant
use option '-std=c99', '-std=gnu99', '-std=c11' or '-std=gnu11' to compile your code
```

if we disable the line 3 in CMakeLists.txt file in the source folder

```cmake
# set_property(TARGET freertos_plus_tcp PROPERTY C_STANDARD 90)
```

the project compiles correctly

**Target**
- Development board: custom development board
- Instruction Set Architecture:  ARM
- IDE and version: vscode
- Toolchain and version:  arm-non-eabi-gcc 10.3.1

```sh
arm-none-eabi-gcc (15:10.3-2021.07-4) 10.3.1 20210621 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

**Host**
- Host OS: GNU/Linux
- Version: Ubuntu 22.04

**To Reproduce**
- Use project ... and configure with ...
- Run on ... and could observe ...

**Expected behavior**

Compile the FreeRTOS-Plus-TCP without any source code changes

**Additional context**

We are creating a custom network driver that leverage FreeRTOS-Plus-TCP V4 features

aggarg, 2024-02-11:
Some fixes were made recently for C90 compliance. Are you using the latest code from main?

wdalmut, 2024-02-11:
@aggarg no, effectively i use the tag V4.0.0 but i got the same list of errors with the main branch

```sh
[build] pico-sdk/src/rp2_common/hardware_gpio/gpio.c:147:5: note: use option '-std=c99', '-std=gnu99', '-std=c11' or '-std=gnu11' to compile your code
[build] pico-sdk/src/rp2_common/hardware_gpio/gpio.c:150:9: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
[build]   150 |         for(uint i=gpio;events8 && i<gpio+8;i++) {
[build]       |         ^~~
```

aggarg, 2024-02-13:
```
[build] pico-sdk/src/rp2_common/hardware_gpio/gpio.c:150:9: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
[build]   150 |         for(uint i=gpio;events8 && i<gpio+8;i++) {
[build]       |  
````
This warning is not from our code but from Pico. Are you setting C90 in your application CMake? Can you share your project that we can use to repro the problem?

wdalmut, 2024-02-13:
@aggarg i'm not but the FreeRTOS-Plus-TCP does it right? This is the line in `CMakeLists.txt` in the `source` folder https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/7a58ae819ce3ca3f79a4e167d4a193da54ba2205/source/CMakeLists.txt#L3 

Here the source file https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/source/CMakeLists.txt#L3

aggarg, 2024-02-14:
The line you mentioned only sets the C90 for freertos_plus_tcp target - so it should not cause failures in Pico code. Did you add more files to freertos_plus_tcp target?

wdalmut, 2024-02-14:
Hi @aggarg you are right during the compilation i have thi `CMakeLists.txt` file

```cmake
set(FREERTOS_HEAP "3")
set(FREERTOS_PORT "GCC_RP2040")

set( FREERTOS_PLUS_TCP_NETWORK_IF "A_CUSTOM_NETWORK_IF" CACHE STRING "" FORCE)
set( FREERTOS_PLUS_TCP_TEST_CONFIGURATION "DISABLE_ALL" CACHE STRING "" FORCE)
set( FREERTOS_PLUS_TCP_BUFFER_ALLOCATION "1" CACHE STRING "" FORCE )
set( FREERTOS_PLUS_TCP_COMPILER "GCC" CACHE STRING "" FORCE )

link_libraries(
    pico_stdlib
    pico_multicore
)

add_subdirectory(FreeRTOS-Config)
add_subdirectory(FreeRTOS-Kernel)
add_subdirectory(FreeRTOS-Plus-TCP/source)
add_subdirectory(FreeRTOS-Plus-TCP/tools)
```

And if i doesn't link the pico library for the kernel i cannot compile for the RP2040 and use the multicore. Do you have any advice to compile the kernel for the rp2040 and the support for the TCP?

aggarg, 2024-02-14:
Nothing seems wrong here at first glance. 

> i can create a dedicated repository as an example for this problem with the pico-sdk to show the problem i just need some time to create it from scratch

Can you do this and share the build instructions also so that we can repro the problem?

tony-josi-aws, 2024-04-01:
Closing this issue as it is not directly related to the +TCP code. 
Feel free to reopen this if you think otherwise.

closed 2024-04-01 by tony-josi-aws