FreeRTOS/FreeRTOS-Plus-TCP #1108: GCC Compilation error

antonio-fr, 2024-02-24:
**Describe the bug**

Compilation issue, related to #958.

error: array subscript is partly outside array bounds.

```
Building C object _deps/freertos_plus_tcp-build/source/CMakeFiles/freertos_plus_tcp.dir/FreeRTOS_DNS_Callback.c.o
In file included from .../build/_deps/freertos_kernel-src/include/./task.h:37,
                 from .../build/_deps/freertos_plus_tcp-src/source/include/FreeRTOS_IP.h:32,
                 from .../build/_deps/freertos_plus_tcp-src/source/include/FreeRTOS_DNS_Callback.h:36,
                 from .../build/_deps/freertos_plus_tcp-src/source/FreeRTOS_DNS_Callback.c:33:
.../build/_deps/freertos_plus_tcp-src/source/FreeRTOS_DNS_Callback.c: In function ‘xDNSDoCallback’:
.../build/_deps/freertos_kernel-src/include/./list.h:241:75: error: array subscript ‘ListItem_t {aka const struct xLIST_ITEM}[0]’ is partly outside array bounds of ‘List_t[1]’ {aka ‘struct xLIST[1]’} [-Werror=array-bounds]
  241 | #define listGET_NEXT( pxListItem )                        ( ( pxListItem )->pxNext )
      |                                                                           ^~
.../build/_deps/freertos_plus_tcp-src/source/FreeRTOS_DNS_Callback.c:72:54: note: in expansion of macro ‘listGET_NEXT’
   72 |             for( pxIterator = ( const ListItem_t * ) listGET_NEXT( xEnd );
      |                                                      ^~~~~~~~~~~~
.../build/_deps/freertos_plus_tcp-src/source/FreeRTOS_DNS_Callback.c:45:19: note: while referencing ‘xCallbackList’
   45 |     static List_t xCallbackList;
      |                   ^~~~~~~~~~~~~
In file included from .../build/_deps/freertos_kernel-src/include/./task.h:37,
                 from .../build/_deps/freertos_plus_tcp-src/source/include/FreeRTOS_IP.h:32,
                 from .../build/_deps/freertos_plus_tcp-src/source/include/FreeRTOS_DNS_Callback.h:36,
                 from .../build/_deps/freertos_plus_tcp-src/source/FreeRTOS_DNS_Callback.c:33:
.../build/_deps/freertos_plus_tcp-src/source/FreeRTOS_DNS_Callback.c: In function ‘vDNSCheckCallBack’:
.../build/_deps/freertos_kernel-src/include/./list.h:241:75: error: array subscript ‘ListItem_t {aka const struct xLIST_ITEM}[0]’ is partly outside array bounds of ‘List_t[1]’ {aka ‘struct xLIST[1]’} [-Werror=array-bounds]
  241 | #define listGET_NEXT( pxListItem )                        ( ( pxListItem )->pxNext )
      |                                                                           ^~
.../build/_deps/freertos_plus_tcp-src/source/FreeRTOS_DNS_Callback.c:205:54: note: in expansion of macro ‘listGET_NEXT’
  205 |             for( pxIterator = ( const ListItem_t * ) listGET_NEXT( xEnd );
      |                                                      ^~~~~~~~~~~~
.../build/_deps/freertos_plus_tcp-src/source/FreeRTOS_DNS_Callback.c:45:19: note: while referencing ‘xCallbackList’
   45 |     static List_t xCallbackList;
      |                   ^~~~~~~~~~~~~
In file included from .../build/_deps/freertos_kernel-src/include/./task.h:37,
                 from .../build/_deps/freertos_plus_tcp-src/source/include/FreeRTOS_IP.h:32,
                 from .../build/_deps/freertos_plus_tcp-src/source/include/FreeRTOS_DNS_Callback.h:36,
                 from .../build/_deps/freertos_plus_tcp-src/source/FreeRTOS_DNS_Callback.c:33:
.../build/_deps/freertos_kernel-src/include/./list.h:241:75: error: array subscript ‘ListItem_t {aka const struct xLIST_ITEM}[0]’ is partly outside array bounds of ‘List_t[1]’ {aka ‘struct xLIST[1]’} [-Werror=array-bounds]
  241 | #define listGET_NEXT( pxListItem )                        ( ( pxListItem )->pxNext )
      |                                                                           ^~
.../build/_deps/freertos_plus_tcp-src/source/FreeRTOS_DNS_Callback.c:240:54: note: in expansion of macro ‘listGET_NEXT’
  240 |             for( pxIterator = ( const ListItem_t * ) listGET_NEXT( xEnd );
      |                                                      ^~~~~~~~~~~~
.../build/_deps/freertos_plus_tcp-src/source/FreeRTOS_DNS_Callback.c:199:16: note: while referencing ‘xTempList’
  199 |         List_t xTempList;
      |                ^~~~~~~~~

```

Looks like exactly as in https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/issues/958#issuecomment-1630222490
I open this issue as we "feel free to create new bug if [we] find any issue with gcc". 

**Target**
- Instruction Set Architecture: AMD64
- Toolchain and version: gcc version 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04)
FREERTOS_PORT=GCC_POSIX

**Host**
- Host OS: Ubuntu
- Version: 22.04.4

**To Reproduce**

Used with cmake CMakeLists :
```
FetchContent_Declare(
  freertos_plus_tcp
  GIT_REPOSITORY https://github.com/FreeRTOS/FreeRTOS-Plus-TCP.git
  GIT_TAG        main
  GIT_SUBMODULES "" 
)
set(FREERTOS_PLUS_TCP_NETWORK_IF "POSIX" CACHE STRING "")
set(FREERTOS_PLUS_FAT_DEV_SUPPORT OFF CACHE BOOL "" FORCE)
set(FREERTOS_PLUS_FAT_PORT "POSIX" CACHE STRING "" FORCE)
set(FREERTOS_PLUS_TCP_TEST_CONFIGURATION "ENABLE_ALL_IPV4_TCP" CACHE INTERNAL "")
FetchContent_MakeAvailable(freertos_plus_tcp)
```

main is commit commit ef14a08 for now

in /build
```
cmake ..
cmake --build .
```

tony-josi-aws, 2024-02-26:
@antonio-fr 

Thanks for taking time to open this bug report.

We are unable to reproduce this issue on GCC 11.4.0, Ubuntu 20.04.6 LTS, when compiled with:

```
     $<$<COMPILE_LANG_AND_ID:C,Clang,GNU>:-Wconversion>
+    $<$<COMPILE_LANG_AND_ID:C,Clang,GNU>:-Warray-bounds>
     $<$<COMPILE_LANG_AND_ID:C,Clang,GNU>:-Wunused-variable>
     $<$<COMPILE_LANG_AND_ID:C,Clang>:-Weverything>
```

```
-- The C compiler identification is GNU 11.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
```

Which project are you testing this with?

antonio-fr, 2024-03-05:
With this modification, it works. it compiles with the "-Os" option.

aggarg, 2024-03-06:
Thank you for reporting back! I am closing this issue. Let us know if you face any other problem.

closed 2024-03-06 by aggarg