FreeRTOS/FreeRTOS-Plus-TCP #1110: [BUG] Freed memory is being used

ATDOR-RIH, 2024-02-27:
**Describe the bug**
This bug releates to #570 which was solved in #707.

The bug was fixed by clearing pPassQueued and pPassAccept.
By calling https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ef14a0871f6ed21c83a97eb176583b5a0cec67e7/source/FreeRTOS_TCP_IP.c#L461
and https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ef14a0871f6ed21c83a97eb176583b5a0cec67e7/source/FreeRTOS_IP_Timers.c#L309
 the socket gets freed, however the parent socket might still have a reference to the socket.
When the user task calls FreeRTOS_accept it will check if the pPassAccept bit is set. If not FreeRTOS_accept will return NULL and everything is fine.
But the memory is freed already, so there is no guarantee that pPassAccept is still cleared.

**Target**
- Development board: [e.g. HiFive11 RevB]
- Instruction Set Architecture: ARM CortexA7
- IDE and version: Eclipse + CMake
- Toolchain and version: arm-none-eabi-gcc 10.3.1

**Host**
- Host OS: Ubuntu
- Version: 18.04

**To Reproduce**
The device acts as web server and the error happened by continuously refreshing the web site using https.
It did not happen when using http.

**Expected behavior**
Do not leave references to freed memory.

**Screenshots**
None

**Wireshark logs**
None

**Additional context**
***Added log lines:***
- logll_debug( BASE_FILE_NAME, "FreeRTOS_socket 0x%x", xReturn ); after https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ef14a0871f6ed21c83a97eb176583b5a0cec67e7/source/FreeRTOS_Sockets.c#L761
- logll_debug( BASE_FILE_NAME, "FreeRTOS_bind 0x%x %li", xSocket, xReturn ); before https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ef14a0871f6ed21c83a97eb176583b5a0cec67e7/source/FreeRTOS_Sockets.c#L1773
- logll_debug( BASE_FILE_NAME, "FreeRTOS_closesocket - 0x%x", xSocket ); after https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ef14a0871f6ed21c83a97eb176583b5a0cec67e7/source/FreeRTOS_Sockets.c#L2019
- logll_debug( BASE_FILE_NAME, "vSocketClose 0x%x", pxSocket ); before https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ef14a0871f6ed21c83a97eb176583b5a0cec67e7/source/FreeRTOS_Sockets.c#L2175
- `if( pxParentSocket->u.xTCP.pxPeerSocket != NULL )
                {
                    logll_debug( BASE_FILE_NAME, "pxClientSocket = pxParentSocket->u.xTCP.pxPeerSocket; 0x%x 0x%x 0x%x",
                            pxClientSocket, pxParentSocket, pxParentSocket->u.xTCP.pxPeerSocket );
                }` before https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ef14a0871f6ed21c83a97eb176583b5a0cec67e7/source/FreeRTOS_Sockets.c#L3892
- `if( pxParentSocket != NULL )
                {
                    logll_debug( BASE_FILE_NAME, "pxClientSocket = pxParentSocket; 0x%x 0x%x", pxClientSocket, pxParentSocket );
                }` before https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ef14a0871f6ed21c83a97eb176583b5a0cec67e7/source/FreeRTOS_Sockets.c#L3896
- logll_debug( BASE_FILE_NAME, "pxClientSocket = NULL;" ); before https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ef14a0871f6ed21c83a97eb176583b5a0cec67e7/source/FreeRTOS_Sockets.c#L3910
- `if( pxClientSocket != NULL && pxClientSocket != FREERTOS_INVALID_SOCKET )
        {
            logll_debug( BASE_FILE_NAME, "FreeRTOS_accept 0x%x 0x%x", pxSocket, pxClientSocket );
        }` before https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ef14a0871f6ed21c83a97eb176583b5a0cec67e7/source/FreeRTOS_Sockets.c#L4059
- logll_debug( BASE_FILE_NAME, "FreeRTOS_listen 0x%x %li", xSocket, xResult ); before https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ef14a0871f6ed21c83a97eb176583b5a0cec67e7/source/FreeRTOS_Sockets.c#L4756
- logll_debug( BASE_FILE_NAME, "vSocketCloseNextTime 0x%x 0x%x", xSocketToClose, pxSocket ); after https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ef14a0871f6ed21c83a97eb176583b5a0cec67e7/source/FreeRTOS_TCP_IP.c#L111
- `else if( pxSocket != NULL )
        {
            logll_debug( BASE_FILE_NAME, "vSocketCloseNextTime set xSocketToClose 0x%x 0x%x", xSocketToClose, pxSocket );
        }` after https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ef14a0871f6ed21c83a97eb176583b5a0cec67e7/source/FreeRTOS_TCP_IP.c#L112
- logll_debug( BASE_FILE_NAME, "xParent = pxSocket->u.xTCP.pxPeerSocket; 0x%x 0x%x 0x%x", xParent, pxSocket, pxSocket->u.xTCP.pxPeerSocket ); after https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ef14a0871f6ed21c83a97eb176583b5a0cec67e7/source/FreeRTOS_TCP_IP.c#L325
- logll_debug( BASE_FILE_NAME, "xParent->u.xTCP.pxPeerSocket = pxSocket; 0x%x 0x%x", xParent, pxSocket ); before https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ef14a0871f6ed21c83a97eb176583b5a0cec67e7/source/FreeRTOS_TCP_IP.c#L345
- logll_debug( BASE_FILE_NAME, "pxSocket->u.xTCP.pxPeerSocket = pxFound; 0x%x 0x%x", pxSocket, pxFound ); before https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ef14a0871f6ed21c83a97eb176583b5a0cec67e7/source/FreeRTOS_TCP_IP.c#L718

***logll_debug*** is a function which writes the given text to a serial interface which is connected to the computer used for debugging.
Format of the log line: (_file name_ provided by macro BASE_FILE_NAME);(_log level_ 1 in case of debug);(_monotonic time_ in ms);(_log message_)

***Log lines where the described bug happens:***
_Creation of socket._
D:FreeRTOS_Sockets.c;1;38081;FreeRTOS_socket 0x7035c548
D:FreeRTOS_TCP_IP.c;1;38113;xParent = pxSocket->u.xTCP.pxPeerSocket; 0x70284130 0x7035c548 0x70284130
D:FreeRTOS_TCP_IP.c;1;38113;xParent->u.xTCP.pxPeerSocket = pxSocket; 0x70284130 0x7035c548

_IP-Task is closing the socket._
D:FreeRTOS_TCP_IP.c;1;38130;vSocketCloseNextTime set xSocketToClose 0x0 0x7035c548
D:FreeRTOS_Sockets.c;1;38130;vSocketClose 0x7035c548
D:FreeRTOS_TCP_IP.c;1;38130;vSocketCloseNextTime 0x7035c548 0x0

_Call FreeRTOS_accept from user task. The memory was not overwritten yet, so prvAcceptWaitClient returns NULL._
D:FreeRTOS_Sockets.c;1;38706;pxClientSocket = pxParentSocket->u.xTCP.pxPeerSocket; 0x0 0x70284130 0x7035c548

***Full logs***
[logs.txt](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/14421120/logs.txt)

***Config***
[ipconfig.txt](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/14421122/ipconfig.txt)

kstribrnAmzn, 2024-02-27:
Thanks for raising this bug. Tagging @shubnil and @tony-josi-aws for review

tony-josi-aws, 2024-03-04:
@ATDOR-RIH 
Thanks for the detailed bug report.
Wondering if you would be able to help with testing if we provided you with a patch for this issue.

ATDOR-RIH, 2024-03-04:
@tony-josi-aws 
Yes, I can help with testing.

tony-josi-aws, 2024-03-05:
@ATDOR-RIH Thanks

In your project, is the macro `ipconfigTCP_HANG_PROTECTION` kept at its default definition? 

If so, could you please define it as 0?:
`#define ipconfigTCP_HANG_PROTECTION                     ( 0 )` 
in your `FreeRTOSIPConfig.h` and try to reproduce the issue?

ATDOR-RIH, 2024-03-06:
I tried it with **#define ipconfigTCP_HANG_PROTECTION                     ( 0 )**.
It is different but there is still an access to freed memory.

D:FreeRTOS_Sockets.c;1;173709;FreeRTOS_socket 0x702b5a60
D:FreeRTOS_TCP_IP.c;1;173722;vSocketCloseNextTime set xSocketToClose 0x0 0x702b5a60
D:FreeRTOS_Sockets.c;1;173722;vSocketClose 0x702b5a60
D:FreeRTOS_TCP_IP.c;1;173722;vSocketCloseNextTime 0x702b5a60 0x0
...
D:FreeRTOS_Sockets.c;1;173963;pxClientSocket = pxParentSocket->u.xTCP.pxPeerSocket; 0x0 0x70283b30 0x702b5a60

tony-josi-aws, 2024-03-07:
@ATDOR-RIH,

Just wondering how its verified that `0x702b5a60` is a freed pointer that is being used.
Its normal to have the malloc return the same pointer that was released while a new socket is being created.

I would try associating each socket creation with a random ID when created and then comparing the IDs when the socket pointers happen to match.

ATDOR-RIH, 2024-03-07:
@tony-josi-aws 
I did what you suggested but instead of a random ID I used the system time of when the socket was created. Therefore I added **ulCreationTime** to _xSOCKET_.
`struct xSOCKET
{
    EventBits_t xEventBits;         /**< The eventbits to keep track of events. */
    EventGroupHandle_t xEventGroup; /**< The event group for this socket. */

    /* Most compilers do like bit-flags */
    struct
    {
        uint32_t bIsIPv6 : 1; /**< Non-zero in case the connection is using IPv6. */
        uint32_t bSomeFlag : 1;
    }
    bits;

    ListItem_t xBoundSocketListItem;       /**< Used to reference the socket from a bound sockets list. */
    TickType_t xReceiveBlockTime;          /**< if recv[to] is called while no data is available, wait this amount of time. Unit in clock-ticks */
    TickType_t xSendBlockTime;             /**< if send[to] is called while there is not enough space to send, wait this amount of time. Unit in clock-ticks */

    IP_Address_t xLocalAddress;            /**< Local IP address */
    uint16_t usLocalPort;                  /**< Local port on this machine */
    uint8_t ucSocketOptions;               /**< Socket options */
    uint8_t ucProtocol;                    /**< choice of FREERTOS_IPPROTO_UDP/TCP */
    uint64_t ulCreationTime;
    #if ( ipconfigSOCKET_HAS_USER_SEMAPHORE == 1 )
        SemaphoreHandle_t pxUserSemaphore; /**< The user semaphore */
    #endif /* ipconfigSOCKET_HAS_USER_SEMAPHORE */
    #if ( ipconfigSOCKET_HAS_USER_WAKE_CALLBACK == 1 )
        SocketWakeupCallback_t pxUserWakeCallback; /**< Pointer to the callback function. */
    #endif /* ipconfigSOCKET_HAS_USER_WAKE_CALLBACK */

    #if ( ipconfigSUPPORT_SELECT_FUNCTION == 1 )
        struct xSOCKET_SET * pxSocketSet; /**< Pointer to the socket set structure */
        EventBits_t xSelectBits;          /**< User may indicate which bits are interesting for this socket. */

        EventBits_t xSocketBits;          /**< These bits indicate the events which have actually occurred.
                                           * They are maintained by the IP-task */
    #endif /* ipconfigSUPPORT_SELECT_FUNCTION */
    struct xNetworkEndPoint * pxEndPoint; /**< The end-point to which the socket is bound. */

    /* This field is only only by the user, and can be accessed with
     * vSocketSetSocketID() / vSocketGetSocketID().
     * All fields of a socket will be cleared by memset() in FreeRTOS_socket().
     */
    void * pvSocketID;

    /* TCP/UDP specific fields: */
    /* Before accessing any member of this structure, it should be confirmed */
    /* that the protocol corresponds with the type of structure */

    union
    {
        IPUDPSocket_t xUDP;     /**< Union member: UDP socket*/
        #if ( ipconfigUSE_TCP == 1 )
            IPTCPSocket_t xTCP; /**< Union member: TCP socket */
        #endif /* ipconfigUSE_TCP */
    }
    u; /**< Union of TCP/UDP socket */
};`

**ulCreationTime** is set in the function _FreeRTOS_socket_.
https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ee730aa2ea5326e09f8ebf60bd676db9fe2bfe72/source/FreeRTOS_Sockets.c#L759

I then added **ulCreationTime** to the logs I mentioned in the description above.
`logll_debug( BASE_FILE_NAME, "FreeRTOS_socket 0x%x %llu", xReturn, xReturn->ulCreationTime );`
`logll_debug( BASE_FILE_NAME, "vSocketClose 0x%x %llu", pxSocket, pxSocket->ulCreationTime );`
`logll_debug( BASE_FILE_NAME, "pxClientSocket = pxParentSocket->u.xTCP.pxPeerSocket; 0x%x 0x%x %llu 0x%x %llu",
                            pxClientSocket, pxParentSocket, pxParentSocket->ulCreationTime, pxParentSocket->u.xTCP.pxPeerSocket,
                            pxParentSocket->u.xTCP.pxPeerSocket->ulCreationTime );`

[... 11 lines cut ...]
You can see that in the log line _pxClientSocket = pxParentSocket->u.xTCP.pxPeerSocket; 0x0 0x70349180 2361 0x7030cdc8 0_ **ulCreationTime** is 0. I assume that the memory was already taken by something else.

I removed `vPortFreeSocket( pxSocket );` from the function _vSocketClose_ and tried again. (log_ipconfigTCP_HANG_PROTECTION_1_do_not_free_memory.txt)
https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ee730aa2ea5326e09f8ebf60bd676db9fe2bfe72/source/FreeRTOS_Sockets.c#L2175

D:FreeRTOS_Sockets.c;1;31698;FreeRTOS_socket 0x7033dab8 31698
D:FreeRTOS_TCP_IP.c;1;31704;xParent = pxSocket->u.xTCP.pxPeerSocket; 0x70349c20 0x7033dab8 31698 0x70349c20 2360
D:FreeRTOS_TCP_IP.c;1;31704;xParent->u.xTCP.pxPeerSocket = pxSocket; 0x70349c20 2360 0x7033dab8 31698
D:FreeRTOS_TCP_IP.c;1;31720;vSocketCloseNextTime set xSocketToClose 0x0 0x7033dab8
D:FreeRTOS_Sockets.c;1;31720;vSocketClose 0x7033dab8 31698
...
D:FreeRTOS_Sockets.c;1;33358;pxClientSocket = pxParentSocket->u.xTCP.pxPeerSocket; 0x0 0x70349c20 2360 0x7033dab8 31698

Then I tried it with ipconfigTCP_HANG_PROTECTION set to 0. (log_ipconfigTCP_HANG_PROTECTION_0_do_not_free_memory.txt)

D:FreeRTOS_Sockets.c;1;210483;FreeRTOS_socket 0x7034eca8 210483
D:FreeRTOS_TCP_IP.c;1;210498;vSocketCloseNextTime set xSocketToClose 0x0 0x7034eca8
D:FreeRTOS_Sockets.c;1;210498;vSocketClose 0x7034eca8 210483
D:FreeRTOS_TCP_IP.c;1;210498;vSocketCloseNextTime 0x7034eca8 0x0
...
D:FreeRTOS_Sockets.c;1;215348;pxClientSocket = pxParentSocket->u.xTCP.pxPeerSocket; 0x0 0x7027d298 2371 0x7034eca8 210483

[log_ipconfigTCP_HANG_PROTECTION_1.txt](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/14523092/log_ipconfigTCP_HANG_PROTECTION_1.txt)
[log_ipconfigTCP_HANG_PROTECTION_1_do_not_free_memory.txt](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/14523093/log_ipconfigTCP_HANG_PROTECTION_1_do_not_free_memory.txt)
[log_ipconfigTCP_HANG_PROTECTION_0_do_not_free_memory.txt](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/14523094/log_ipconfigTCP_HANG_PROTECTION_0_do_not_free_memory.txt)

shubnil, 2024-03-21:
Hi @ATDOR-RIH,
We were analyzing the logs and have some observations:

Data:
------
Cycle 1:
FreeRTOS_Sockets.c;1;31698;FreeRTOS_socket **0x7033dab8 31698**
FreeRTOS_TCP_IP.c;1;31704;xParent = pxSocket->u.xTCP.pxPeerSocket; 0x70349c20 0x7033dab8 31698 0x70349c20 2360
FreeRTOS_TCP_IP.c;1;31704;xParent->u.xTCP.pxPeerSocket = pxSocket; 0x70349c20 2360 **0x7033dab8 31698**
FreeRTOS_TCP_IP.c;1;31720;vSocketCloseNextTime set xSocketToClose 0x0 0x7033dab8
FreeRTOS_Sockets.c;1;31720;vSocketClose 0x7033dab8 31698
FreeRTOS_TCP_IP.c;1;31720;vSocketCloseNextTime 0x7033dab8 0x0

Cycle 2:
FreeRTOS_Sockets.c;1;31853;FreeRTOS_socket 0x702f3cb8 31853
FreeRTOS_TCP_IP.c;1;31859;xParent = pxSocket->u.xTCP.pxPeerSocket; 0x70349c20 0x702f3cb8 31853 0x70349c20 2360
FreeRTOS_TCP_IP.c;1;31869;vSocketCloseNextTime set xSocketToClose 0x0 0x702f3cb8
FreeRTOS_Sockets.c;1;31869;vSocketClose 0x702f3cb8 31853
FreeRTOS_TCP_IP.c;1;31869;vSocketCloseNextTime 0x702f3cb8 0x0

Cycle 3:
FreeRTOS_Sockets.c;1;32004;FreeRTOS_socket 0x7025edf0 32004
FreeRTOS_TCP_IP.c;1;32010;xParent = pxSocket->u.xTCP.pxPeerSocket; 0x70349c20 0x7025edf0 32004 0x70349c20 2360
FreeRTOS_TCP_IP.c;1;32020;vSocketCloseNextTime set xSocketToClose 0x0 0x7025edf0
FreeRTOS_Sockets.c;1;32020;vSocketClose 0x7025edf0 32004
FreeRTOS_TCP_IP.c;1;32020;vSocketCloseNextTime 0x7025edf0 0x0

Cycle 4:
FreeRTOS_Sockets.c;1;32214;FreeRTOS_socket 0x702843d0 32214
FreeRTOS_TCP_IP.c;1;32220;xParent = pxSocket->u.xTCP.pxPeerSocket; 0x70349c20 0x702843d0 32214 0x70349c20 2360
FreeRTOS_TCP_IP.c;1;32231;vSocketCloseNextTime set xSocketToClose 0x0 0x702843d0
FreeRTOS_Sockets.c;1;32231;vSocketClose 0x702843d0 32214
FreeRTOS_TCP_IP.c;1;32231;vSocketCloseNextTime 0x702843d0 0x0

Cycle 5:
FreeRTOS_Sockets.c;1;32779;FreeRTOS_socket 0x7033ace0 32779
FreeRTOS_TCP_IP.c;1;32791;xParent = pxSocket->u.xTCP.pxPeerSocket; 0x70349c20 0x7033ace0 32779 0x70349c20 2360
FreeRTOS_Sockets.c;1;33358;FreeRTOS_closesocket - 0x7027d2d8 31569
FreeRTOS_Sockets.c;1;33358;pxClientSocket = pxParentSocket->u.xTCP.pxPeerSocket; 0x0 0x70349c20 2360 **0x7033dab8 31698**
FreeRTOS_Sockets.c;1;33358;pxClientSocket = NULL;
----------------------------------
Observation 1: The socket buffer with address "0x7033dab8" seem to appear with an older timestamp. Now this can be either because of one of the following reasons:

1. Stale buffer is still present
2. The "ulCreationTime" field is not cleared properly which can happen if the print statement is before the socket memory is memset to 0.

There is less possibility for a to happen because of the following prints:
FreeRTOS_Sockets.c;1;31720;vSocketClose 0x7033dab8 31698
FreeRTOS_TCP_IP.c;1;31720;vSocketCloseNextTime 0x7033dab8 0x0

Can you please confirm if point no b is not the case.

Observation 2: 
Sometimes there seems to be a late close call from the application like the print "FreeRTOS_Sockets.c;1;33358;FreeRTOS_closesocket - 0x7027d2d8 31569"
Can you please ensure if the application has the right priority to maintain a steady state between opening and closing of sockets.

Will it be ok for you to share the code with the above prints. This will help us better co-related the print sequence and the code sequence.

Thanks,
Shub

htibosch, 2024-03-23:
I would like to propose the following change in `vTCPStateChange()`:

~~~diff
 FreeRTOS_printf( ( "vTCPStateChange: Closing socket\n" ) );
 
 if( pxSocket->u.xTCP.bits.bReuseSocket == pdFALSE_UNSIGNED )
 {
     configASSERT( xIsCallingFromIPTask() != pdFALSE );
+ 
+    if( ( xParent != NULL ) &&
+        ( xParent->u.xTCP.pxPeerSocket == pxSocket ) )
+    {
+        xParent->u.xTCP.pxPeerSocket = NULL;
+    }
 
     vSocketCloseNextTime( pxSocket );
 }
~~~
So we'll clear the field `pxPeerSocket` of the parent socket before the child socket is deleted.

ATDOR-RIH, 2024-03-25:
Hi @shubnil and @htibosch 

attached you will find the following files:
- changes.patch: Patch file with the changes I made based on V4.0.0. (relative path: source)
- changes_suggestion_htibosch.patch: Patch file with the changes suggested by @htibosch.
- logs_suggestion_htibosch.txt: Logs with the changes suggested by @htibosch. (Only a small snippet.)

[changes.patch](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/14747094/changes.patch)
[changes_suggestion_htibosch.patch](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/14747095/changes_suggestion_htibosch.patch)
[logs_suggestion_htibosch.txt](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/14747096/logs_suggestion_htibosch.txt)

The suggested changes do not solve the issue.
D:FreeRTOS_TCP_IP.c;1;658742;vSocketCloseNextTime xParent 0x7027b4d8 pxSocket 0x7027b4d8 0x0
D:FreeRTOS_TCP_IP.c;1;658742;vSocketCloseNextTime set xSocketToClose 0x0 0x7027b4d8
D:FreeRTOS_Sockets.c;1;658742;vSocketClose 0x7027b4d8 658696
D:FreeRTOS_TCP_IP.c;1;658742;vSocketCloseNextTime 0x7027b4d8 0x0
...
D:FreeRTOS_Sockets.c;1;660547;pxClientSocket = pxParentSocket->u.xTCP.pxPeerSocket; 0x0 0x702f5068 2247 0x7027b4d8 658696

I added this log "vSocketCloseNextTime xParent 0x7027b4d8 pxSocket 0x7027b4d8 0x0" right before the suggested changes.
Here you can see that xParent is the same as pxSocket and pxSocket->u.xTCP.pxPeerSocket is NULL.

Regarding priorities:
The IP-Task has the same priority as the client tasks.

htibosch, 2024-03-26:
@ATDOR-RIH wrote:
> Regarding priorities:
> The IP-Task has the same priority as the client tasks.

**Normally** we recommend to use these priorities:
- highest: the EMAC / Ethernet task
- high: the IP-task with `ipconfigIP_TASK_PRIORITY`
- normal: all tasks that make use of TCP/IP

Tasks that have nothing to do with TCP/IP are free to choose a priority.

I will comment `pxPeerSocket` in a later post.

ATDOR-RIH, 2024-03-26:
Here are the extended logs:
[logs_suggestion_htibosch_extended.txt](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/14754132/logs_suggestion_htibosch_extended.txt)

At timestamp 642815 the socket (created at timestamp 639023) is assigned as peer socket of the parent.

Is it possible to set the peer socket of pxFound in xTCPCheckNewClient or is this a bad idea?
                if( ( pxFound->ucProtocol == ( uint8_t ) FREERTOS_IPPROTO_TCP ) && ( pxFound->u.xTCP.bits.bPassAccept != pdFALSE_UNSIGNED ) )
                {
                    logll_debug( BASE_FILE_NAME, "pxSocket->u.xTCP.pxPeerSocket = pxFound; 0x%x %llu 0x%x %llu", pxSocket,
                            pxSocket->ulCreationTime, pxFound, pxFound->ulCreationTime );
                    pxSocket->u.xTCP.pxPeerSocket = pxFound;
                    **pxFound->u.xTCP.pxPeerSocket = pxSocket;**
                    FreeRTOS_debug_printf( ( "xTCPCheckNewClient[0]: client on port %u\n", pxSocket->usLocalPort ) );
                    xResult = pdTRUE;
                    break;
                }

htibosch, 2024-03-26:
When I look at your latest logs, I see lots of calls to `FreeRTOS_socket()`, and all those sockets end up being closed in `vSocketCloseNextTime()`?
That function is only called as long as a socket has not been passed to the application.
Can you see for what reason `vSocketCloseNextTime()` is called? Does it reach a timeout? Does it get a bad response from the peer?
Could you create and send a PCAP with a few TCP connections? Please zip and attach it.

Another thing: could you run the test while the application has a lower priority than `ipconfigIP_TASK_PRIORITY`?

shubnil, 2024-03-27:
Hi @ATDOR-RIH ,
Please check once the application priority is the lowest. A large number of "FreeRTOS_socket()" call seem to point to the fact that the application priority is not the lowest in the system.

ATDOR-RIH, 2024-03-28:
I think vSocketCloseNextTime() is called because of a RST, but I am not sure.

Changes:
- Added logs at every call to vTCPStateChange with eCLOSED or eCLOSE_WAIT.
- Added pxParentSocket to struct xSOCKET and log it when vSocketCloseNextTime is called.

[pcap_and_changed_priorities.zip](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/14792586/pcap_and_changed_priorities.zip)

pcap_and_changed_priorities.zip contains the following folders and files:
- original_priorities: Logs and Pcap file of a run with the priorities as they were.
- changed_priorities: Logs and Pcap file of a run with changed priorities.
- changes.patch: changes based from version 4.0.0
- changes_from_last_time.patch: changes based from the last patch I sent

folder contents:
- log.txt
- original_priorities.pcap/changed_priorities.pcap
- info.txt: Timestamps and used priorities.

Changing the priorities did not help.
I tried to run the test for around 90 seconds.
With the changed priorities I could not make it past 70 seconds without receiving a hard fault/bus fault.
I also get hard faults with the original priorities but not that frequently.

htibosch, 2024-03-30:
Thanks for these tests. This gives a new insight: the client (browser) manages to get a connection, and then immediately it is closing it. This lasts like 15 ms.

Here is an example:

[image]

So it's not a timeout, it is not cause by an unexpected RST packet. The client is closing the connection immediately for some reason.

I will think about this and come back to it.

htibosch, 2024-03-31:
I made a simulation and get the same pattern of events:

[image]

When debug printf enabled, I see this logging:
~~~
/* Server side: */
       IP   SS  MS  US
x.x.x.104   47.746.997 [IP-Task] Gain: Socket 23 now has 1 / 4 child
x.x.x.104   47.747.053 [IP-Task] prvSocketSetMSS: 1460 bytes for 192.168.2.107 port 50709
x.x.x.104   47.747.125 [IP-Task] prvWinScaleFactor: uxRxWinSize 1 MSS 1460 Factor 0
x.x.x.104   47.747.673 [IP-Task] TCP: passive 23 => 107.2.168.192 port 50709 set ESTAB (scaling 1)
x.x.x.104   47.747.785 [SvrWork] xTelnetRead: new client from 192.168.2.107:50709
x.x.x.104   47.748.433 [SvrWork] xTelnetRead: client 20014ad0 disconnected (rc -128)
x.x.x.104   47.748.495 [IP-Task] Lost: Socket 23 now has 0 / 4 children
x.x.x.104   47.748.578 [IP-Task] FreeRTOS_closesocket[0.0.0.0 port 23 to 192.168.2.107 port 50709]: buffers 14 socks 1

/* Clients side: */
x.x.x.107     78.643.976 [SvrWork] FreeRTOS_connect: 50709 to 192.168.2.104:23
x.x.x.107     78.644.059 [SvrWork] Socket 50709 -> [192.168.2.104]:23 State eCLOSED->eCONNECT_SYN
x.x.x.107     78.644.174 [IP-task] prvSocketSetMSS: 1460 bytes for 192.168.2.104 port 23
x.x.x.107     78.644.255 [IP-task] prvWinScaleFactor: uxRxWinSize 1 MSS 1460 Factor 0
x.x.x.107     78.644.347 [IP-task] Connect[192.168.2.104:23]: next timeout 1: 3000 ms
x.x.x.107     78.644.714 [IP-task] TCP: active 50709 => 104.2.168.192 port 23 set ESTAB (scaling 1)
x.x.x.107     78.644.927 [IP-task] Socket 50709 -> [192.168.2.104]:23 State eCONNECT_SYN->eESTABLISHED
x.x.x.107     78.645.157 [SvrWork] FreeRTOS_connect returns 0
x.x.x.107     78.645.251 [IP-task] Socket 50709 -> [192.168.2.104]:23 State eESTABLISHED->eFIN_WAIT_1
x.x.x.107     78.645.582 [IP-task] vTCPStateChange: Closing socket (Queued 0, Accept 0 Reuse 0)
x.x.x.107     78.645.662 [IP-task] Socket 50709 -> [192.168.2.104]:23 State eFIN_WAIT_1->eCLOSE_WAIT
x.x.x.107     78.645.994 [IP-task] FreeRTOS_closesocket[0.0.0.0 port 50709 to 192.168.2.104 port 23]: buffers 12 socks 3
~~~

I made some extra logging in FreeRTOS_TCP_IP.c: `vTCPStateChange()`:

~~~c
if( ( eTCPState == eCLOSED ) ||
    ( eTCPState == eCLOSE_WAIT ) )
{
    /* Socket goes to status eCLOSED because of a RST.
     * When nobody owns the socket yet, delete it. */
    FreeRTOS_printf( ( "vTCPStateChange: Closing socket (Queued %d, Accept %d Reuse %d)\n",
        pxSocket->u.xTCP.bits.bPassQueued,
        pxSocket->u.xTCP.bits.bPassAccept,
        pxSocket->u.xTCP.bits.bReuseSocket ) );
    ...
}
~~~

Would you mind to send me the same type of logging, i.e. by enabling the following in your FreeRTOSIPConfig.h :

~~~
/* Your printing routine that doesn't need Ethernet. */
extern void logll_debug( const char *pcFormatString, ... );

#define ipconfigHAS_DEBUG_PRINTF    1

#if( ipconfigHAS_DEBUG_PRINTF == 1 )
    #define FreeRTOS_debug_printf( X )    ( void ) logll_debug X
#endif

#define ipconfigHAS_PRINTF            1

#if( ipconfigHAS_PRINTF == 1 )
    #define FreeRTOS_printf(X)            ( void ) logll_debug X
#endif
~~~

I can not replicate the problem. It would help to understand the order of events. 
Who is calling `vTCPStateChange()` with the parameter `eCLOSED` or `eCLOSE_WAIT`?

htibosch, 2024-04-01:
Hi @ATDOR-RIH, after attempts to reproduce your problem, here is a possible solution:

[Source_hein.zip](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/14824893/Source_hein.zip)

Please merge them into you repo using a source compare program.

It looks like it is possible that the field `pxPeerSocket` is still set when the application calls `FreeRTOS_accept()`.

I'm not claiming that all is solved, but I would be very curious if you can show the logging produced by the TCP/IP stack.

ATDOR-RIH, 2024-04-02:
Hi @htibosch, logs.zip contains three different logs.

- additional_logs: Here I just activated the logs you suggested in the previous comment. I also removed some of the logs I added.
- changes_from_Source_hein: Here I applied the changes from Source_hein.zip. I am working on 4.0.0 and I think the files you provided are based on main. I compared the changes to main and copied them to the version I am currently on. The changed files are also in the folder. I hope I did not forget anything. I got a bus fault with this version.
- change_condition: Here I changed a condition in FreeRTOS_TCP_IP.c (file is included). This on also resulted in a bus fault.

[logs.zip](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/14839282/logs.zip)

htibosch, 2024-04-05:
There are too many differences between your and my source code. Please find my complete directory "source":

[complete_sources_hein.zip](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/14881029/complete_sources_hein.zip)

except for the directory "portable".

I tested this version with real hardware, and the problem looks "solved".
Could you please try it and send me the logging output?

Thanks for you patience and corporation, Hein

ATDOR-RIH, 2024-04-11:
Hi @htibosch, sorry for the delay.

There was no fault but freed memory is still being accessed. But it seems to happen a lot less.

Here are the logs of three runs as well as the source files I used. I kept the logs I added but otherwise it should be the same as your files.

[logs.zip](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/14945471/logs.zip)

htibosch, 2024-04-14:
Can you please point out where in the logfile you can see that freed memory is being accessed?
Which statement/function?

ATDOR-RIH, 2024-04-15:
This is from log1.txt. It is similar for the other log files.

Here **0x703c7018** is set to be closed the next time in FreeRTOS_TCP_IP.c vTCPStateChange(FreeRTOS_Socket_t * pxSocket, enum eTCP_STATE eTCPState). **0x702e8920** is the parent of this socket.

> D:FreeRTOS_TCP_IP.c;1;241165;vSocketCloseNextTime xParent 0x703c7018 pxSocket 0x703c7018 0x0 0x702e8920

Here the socket is closed.

> D:FreeRTOS_Sockets.c;1;241165;vSocketClose 0x703c7018 240696

Here **0x703c7018** is still the peer socket of **0x702e8920** and is accessed here: **if( pxClientSocket->u.xTCP.bits.bPassAccept != pdFALSE_UNSIGNED )** in FreeRTOS_Sockets.c prvAcceptWaitClient(FreeRTOS_Socket_t * pxParentSocket, struct freertos_sockaddr * pxAddress, socklen_t * pxAddressLength)

> D:FreeRTOS_Sockets.c;1;243123;pxClientSocket = pxParentSocket->u.xTCP.pxPeerSocket; 0x0 0x702e8920 2368 0x703c7018 240696

pxClientSocket (0x703c7018) is already free when accessing pxClientSocket->u.xTCP.bits.bPassAccept. In this case the memory is still free and still has the correct value, but there is no guarantee that it does not get allocated and overwritten.

htibosch, 2024-04-15:
How strange, both with your source and my sources, I can not replicate the problem you mention.

As you know:

- for a TCP server, `pxPeerSocket` points to a TCP client
- for a TCP client, `pxPeerSocket` points to its TCP server

In your case, ``server->pxPeerSocket` doesn't get cleared.

Please find a proposal to further investigate this: [FreeRTOS_TCP_IP_hein.zip](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/14983773/FreeRTOS_TCP_IP_hein.zip)

I added a function:
~~~
BaseType_t vTCPRemoveTCPChild( const FreeRTOS_Socket_t * pxChildSocket )
~~~

It will iterate through all sockets to find a socket the is the parent of `pxChildSocket`. When found, the reference will be clear ( set to NULL ).

ATDOR-RIH, 2024-04-19:
Hi @htibosch,
yes with this change it is working.

[log.txt](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/15041398/log.txt)

server->pxPeerSocket doesn't get cleared because _client->pxPeerSocket_ is NULL at some point when _vSocketCloseNextTime_ is called.
Log: vSocketCloseNextTime xParent
In this log line you can see that _client->pxPeerSocket_ is NULL but _client->pxParentSocket_ has the the address of the server.
But I don't know why _client->pxPeerSocket_ is NULL.
I thought it was because of this:
```
/* Don't need to access the parent socket anymore, so the
* reference 'pxPeerSocket' may be cleared. */
pxSocket->u.xTCP.pxPeerSocket = NULL;
```
But this is already removed in the source I am using and it is still happening.

I will try to find out where _client->pxPeerSocket_ is set to NULL or if it is not set at all. But this might take some time.

htibosch, 2024-04-20:
That is good news!

htibosch, 2024-04-22:
Could it be that you do not have `ipconfigTCP_HANG_PROTECTION` defined?

In that case, see FreeRTOS_TCP_State_Handling.c, around line 995 :

 ~~~diff
     BaseType_t prvTCPSocketCopy( FreeRTOS_Socket_t * pxNewSocket,
                                  FreeRTOS_Socket_t * pxSocket )
     {
     ...
     #if ( ipconfigTCP_HANG_PROTECTION == 1 )
         {
             /* Only when there is anti-hanging protection, a socket may become an
              * orphan temporarily.  Once this socket is really connected, the owner of
              * the server socket will be notified. */
 
             /* When bPassQueued is true, the socket is an orphan until it gets
              * connected. */
             pxNewSocket->u.xTCP.bits.bPassQueued = pdTRUE_UNSIGNED;
             pxNewSocket->u.xTCP.pxPeerSocket = pxSocket;
         }
         #else
         {
             /* A reference to the new socket may be stored and the socket is marked
              * as 'passable'. */
 
             /* When bPassAccept is true, this socket may be returned in a call to
              * accept(). */
             pxNewSocket->u.xTCP.bits.bPassAccept = pdTRUE_UNSIGNED;
+            /* Assign the parent 'pxSocket' to the child socket. */
+            pxNewSocket->u.xTCP.pxPeerSocket = pxSocket;
 
             if( pxSocket->u.xTCP.pxPeerSocket == NULL )
             {
                 pxSocket->u.xTCP.pxPeerSocket = pxNewSocket;
             }
         }
         #endif /* if ( ipconfigTCP_HANG_PROTECTION == 1 ) */
~~~

Could you test the above patch while disabling the new function `vTCPRemoveTCPChild()`? This only makes sense if you have `ipconfigTCP_HANG_PROTECTION = 0`.

Thanks

ATDOR-RIH, 2024-04-30:
Hi @htibosch,

yes you are right. ipconfigTCP_HANG_PROTECTION is 0.
I even tested with with ipconfigTCP_HANG_PROTECTION set to 0 and 1 on March 7 and then forgot about it somewhen.

[logs.zip](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/files/15165578/logs.zip)

logs.zip contains logs with ipconfigTCP_HANG_PROTECTION set to 1 for different versions.
- log_log_disabled_vTCPRemoveTCPChild.txt: The changes you provided on April 15 with vTCPRemoveTCPChild() disabled.
- log_complete_sources_hein.txt: The version you provided on April 5.
- log_sources_hein.txt: The changes you provided on April 1.
- log_march_26.txt: The version I used before getting your changes, where I changed the priorities according to your suggestion on March 26.

With the version from March 26 the problem still exists with ipconfigTCP_HANG_PROTECTION set to 1.
Since the changes from April 1 the problem is solved, if ipconfigTCP_HANG_PROTECTION is set to 1.

Thanks for your help

tony-josi-aws, 2024-06-04:
@ATDOR-RIH 
Closing this issue as PR #1148  should be solving the issue.
Please feel free to reopen it if you think otherwise.

closed 2024-06-04 by tony-josi-aws