FreeRTOS/FreeRTOS-Plus-TCP #1128: Compile error with DNS cache disabled [BUG]

evpopov, 2024-04-02:
Pretty much what the title says:

**To Reproduce**
```
#define ipconfigUSE_DNS                   ipconfigENABLE
#define ipconfigUSE_DNS_CACHE             ipconfigDISABLE 
```
I get an error: `struct <anonymous>' has no member named 'ucName'`

Skptak, 2024-04-04:
Hey @evpopov, I opened #1131 to fix this issue. I've messaged @tony-josi-aws about this as well.

Skptak, 2024-04-09:
> My first quick-and-dirty fix would have been different than what you did in https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/1131

Hey @evpopov, not an issue that you'd have a different fix for this than I did. In the future feel free to open a PR up with your proposed changes, the team can always discuss the benefits and downsides of an approach in the PR!

> Just to let you know, I did apply your changes from https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/1131 to my test machine where I have multiple DUTs that resolve each other by name and everything seems to be running fine.

I'm glad to know that the fix worked for you! Thanks for testing it on your local setup as well!

> As a quick test, I completely removed those fields and all code associated with them (which is actually just the 3 lines you touched in https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/1131 )

I believe that `pxAddrInfo->xPrivateStorage.ucName` holds the string name of the DNS lookup. @tony-josi-aws or @htibosch could probably validate this. Where removing these lines should still work, just potentially cause a DNS lookup to need to occur when connecting to the address.

tony-josi-aws, 2024-04-12:
> Could you tell me what `pxAddrInfo->ai_canonname` and `pxAddrInfo->xPrivateStorage.ucName` are intended to be used for?

I believe during the DNS cache lookup, for example, in the `FreeRTOS_ProcessDNSCache` if a valid `ppxAddressInfo` is provided, it copies all the DNS entries (for example, if `ipconfigDNS_CACHE_ADDRESSES_PER_ENTRY` > 1) along with `ai_canonname`  as a linked `struct addrinfo`. Currently, `FreeRTOS_dnslookup` or `FreeRTOS_dnslookup6` doesn't seem to use it.

evpopov, 2024-04-12:
Thanks for the explanation @tony-josi-aws

closed 2024-04-12 by evpopov