FreeRTOS/FreeRTOS-Plus-TCP #1133: ipconfigCOMPATIBLE_WITH_SINGLE & ipconfigIPv4_BACKWARD_COMPATIBLE are used inconsistently

HTRamsey, 2024-04-05:
I assume ipconfigIPv4_BACKWARD_COMPATIBLE is for just IPv4/IPv6 & ipconfigCOMPATIBLE_WITH_SINGLE is only relevant for FreeRTOS_Routing.c. However, some Network Interfaces use ipconfigIPv4_BACKWARD_COMPATIBLE and others use ipconfigIPv4_BACKWARD_COMPATIBLE when guarding pxFillInterfaceDescriptor.

Skptak, 2024-04-06:
I believe `ipconfigCOMPATIBLE_WITH_SINGLE` is supposed to be a different config for when only allowing a single network interface. I believe the idea is that this is separate from the changes involved in `ipconfigIPv4_BACKWARD_COMPATIBLE` at certain places. @htibosch or @tony-josi-aws may have more accurate information.

HTRamsey, 2024-04-06:
Here is an example of what I am referring to:
[NXP1060](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ebf63bd3da2a9d86ddc41da87fea247b5fde8eeb/source/portable/NetworkInterface/NXP1060/NetworkInterface.c#L272)
[STM32Hxx](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/ebf63bd3da2a9d86ddc41da87fea247b5fde8eeb/source/portable/NetworkInterface/STM32Hxx/NetworkInterface.c#L538)

tony-josi-aws, 2024-04-08:
Created a PR #1134 to fix the issue

closed 2024-04-08 by HTRamsey