FreeRTOS/FreeRTOS-Plus-TCP #1149: [BUG] Allocation failure in `prvCreateSectors` is silently ignored, causing unexpected packet send failures later on

hlef, 2024-05-31:
**Describe the bug**
When the allocation of `xTCPSegments`  in `prvCreateSectors` fails ([here](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/source/FreeRTOS_TCP_WIN.c#L429)), an error code is returned. However this error code is ignored by the caller `vTCPWindowCreate` ([here](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/source/FreeRTOS_TCP_WIN.c#L793) - see the explicit `void` cast). The result is a network stack that seems to correctly initialize, but fails to send packets through `FreeRTOS_send` later on.

We are encountering this issue in a setting where the network stack is initialized late, and not enough memory is available to satisfy the relatively large allocation in `prvCreateSectors`.

This is a very tricky issue to debug, because not all packets are impacted by this allocation failure. In our case, we perform a TLS handshake: the TCP handshake goes through, but the TLS client hello does not send because of this problem.

**Target**
Any, AFAIK this is not a target-specific bug.

**Host**
Same.

**To Reproduce**
- Configure with `ipconfigUSE_TCP_WIN`.
- Run in a context where the allocation of `xTCPSegments`  in `prvCreateSectors` fails ([here](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/source/FreeRTOS_TCP_WIN.c#L429))
- Observe that the network stack initialization succeeds but later fails to send packets through `FreeRTOS_send` later on

**Expected behavior**
The failure of `prvCreateSectors` should be noticed by `vTCPWindowCreate`, and further propagated up the call stack. This should ultimately result in the network stack failing to initialize.

**Screenshots**
N/A.

**Wireshark logs**
In Wireshark, we observe that the TCP handshake goes through, but the client hello packet of our TLS handshake is not sent.

**Additional context**
N/A.

tony-josi-aws, 2024-06-02:
Hi @hlef,

Thanks for reporting this issue.
Please let us know if you prefer us to make a PR or if you prefer to create one since you have already figured out the cause of this bug.

hlef, 2024-06-05:
Hi @tony-josi-aws I would prefer if you could handle the PR as this may take a while to reach the top of my TODO list :slightly_smiling_face:

tony-josi-aws, 2024-06-17:
@hlef 

Closing this ticket as PR #1152 should fix this issue.

closed 2024-06-17 by tony-josi-aws