FreeRTOS/FreeRTOS-Plus-TCP #1150: [Feature Request] Support EUI-64 or similar for global unicast IPv6 addresses derived from RA

ljanyst, 2024-05-31:
**Is your feature request related to a problem? Please describe.**
When router advertisements are used as a means of IPv6 address configuration, the IP stack generates a random host id, appends it to the prefix, and sends a neighbor solicitation to rule out a collision: https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/source/FreeRTOS_RA.c#L579 This works fine if you assume that the device using the stack will be a client. However, when the device needs to be a server, a stable address is needed. Normally, EUI-64 or a similar method could be used for generation of such address.

**Describe the solution you'd like**
Please add a way to derive stable addresses from router advertisements.

**Describe alternatives you've considered**
I am forced to assign my addresses statically.

paulbartell, 2024-06-03:
Thanks for the feature request. I'll forward it to the team!

shubnil, 2024-06-13:
Hi @ljanyst , Thanks for bringing this out.
Yes, we do not support generation of IPv6 address in router advertisements. The main reason being:
1. We do not support Ipv6 extension headers as yet and this being closely linked to extension headers, the plan was to do them together
2. Secondly, as most of our customer devices run in client mode, there has not been an explicit request so far.

However, we would like to have the functionality.
Would you be interested in contributing the same by creating a PR.
We would be more than happy to help you merge the same as soon as possible.

ljanyst, 2024-06-18:
Sure, I will take a look.