FreeRTOS/FreeRTOS-Plus-TCP #1167: Compilation error because of 'xMDNS_MACAddressIPv6'

htibosch, 2024-07-07:
**Describe the problem**

I am proposing the following change :
~~~diff
     const MACAddress_t xLLMNR_MacAddress;
     const MACAddress_t xLLMNR_MacAddressIPv6;
     const MACAddress_t xMDNS_MacAddress;
-    const MACAddress_t xMDNS_MACAddressIPv6;
+    const MACAddress_t xMDNS_MacAddressIPv6;
~~~
`xMDNS_MACAddressIPv6` is the only identifier who writes `_MAC` with capitals.

- **Target**
- **Host**

Any target or host

- Use project ... and configure with ...

Compile a project using either `STM32Hxx`, `Zynq`, or `DriverSAM`.
Enable `ipconfigUSE_MDNS` and `ipconfigUSE_IPv6`.

I think that `_MACAddress` with capitals is more correct, because MAC is an abbreviation. However, `_MacAddress` is maybe easier to read.

It would be nice to have a solution that is downward compatible, such as:
~~~c
    const MACAddress_t xMDNS_MacAddressIPv6;
    #define xMDNS_MACAddressIPv6  xMDNS_MacAddressIPv6
~~~

Note that user-written modules are free to use `xMDNS_MACAddressIPv6`, and we don't want to break existing code.

ActoryOu, 2024-07-10:
Hi @htibosch,
Yeah, I agree that we should align the naming to `xMDNS_MacAddressIPv6`.
I would raise a PR for this.

Thank you!

ActoryOu, 2024-07-11:
Closing because fix was merged.

closed 2024-07-11 by ActoryOu