FreeRTOS/FreeRTOS-Plus-TCP #1168: [Feature Request] ESP32 port implementation for both Ethernet and Wi-Fi

ryan4xen, 2024-07-08:
I am currently porting the TCP stack to work with the esp-idf as an alternative to the existing lwIP. Though during the implementation I found that only Wi-Fi was actually implemented, but using `_Eth_` naming. At this stage I am only personally doing a Wi-Fi implementation, though would be good to clean this up.

For example:
```
static BaseType_t xESP32_Eth_NetworkInterfaceInitialise( NetworkInterface_t * pxInterface )
{
    static BaseType_t xMACAdrInitialized = pdFALSE;
    uint8_t ucMACAddress[ ipMAC_ADDRESS_LENGTH_BYTES ];

    if( xInterfaceState == INTERFACE_UP )
    {
        if( xMACAdrInitialized == pdFALSE )
        {
            esp_wifi_get_mac( ESP_IF_WIFI_STA, ucMACAddress );
            FreeRTOS_UpdateMACAddress( ucMACAddress );
            xMACAdrInitialized = pdTRUE;
        }

        return pdTRUE;
    }

    return pdFALSE;
}
```

Unless I am missing something I believe someone originally intended to implement the Ethernet driver, but ended up with a Wi-Fi one instead.

htibosch, 2024-07-08:
The +TCP driver for ESP32 in [portable/NetworkInterface/esp32](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/main/source/portable/NetworkInterface/esp32), is indeed only for WiFi.

These seem to be the official driver sources on [github.com/espressif](https://github.com/espressif/esp-eth-drivers).

If you decide to create a ESP32 Ethernet driver for +TCP, you can either:

- Ask for support [on the FreeRTOS forum](https://forums.freertos.org/c/libraries)
- Create a pull request for it on [FreeRTOS+TCP/pulls](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pulls).

It shouldn't be too difficult, assuming that the espressif driver is well tested.

moninom1, 2024-07-22:
HI @ryan4xen 
Are you planning on creating a pull request as suggested by hein?